adafruit/Adafruit-Fingerprint-Sensor-Library #9: Template comparison

nonthanut, 2016-02-09:
I notice I could extract a fingerprint's template from the library. I wonder whether there is anyway to upload the template to my server where I can compute the matching score.

(As the FP sensor could store a limited number of fingerprints, I plan to keep a database for all the templates on the server instead. The issue now is how to match a template at the server.)

kalyanisoholkar, 2016-03-17:
I have the same problem. I have to save the template in a mysql database and then again retrieve it to match. I extracted the template now i need to send this template for matching,but i dont know hwere to send this template for matching..Please help..

information-security, 2017-04-21:
Any hints on this? a SDK is a must have for this product. If anyone has any idea what is the algorithm I can do the coding part.

alphashuro, 2017-06-07:
has anyone figured this out yet? i'm also trying to do this

sqwerty13, 2017-07-19:
I have this problem too

drichardcarl, 2017-10-12:
Does anyone know how to match fingerprint templates when you have them ? I can extract the templates generated after the enrollment. In my case, after enrollment, I don't store the template to the internal memory of the device but I store the template to my computer.

Obaid-ur-Rehman, 2017-10-31:
Please post a sample code to extract template.
I see there is no function to extract template in the library.
Thanks

drichardcarl, 2017-11-02:
`#include <Adafruit_Fingerprint.h>
#define mySerial Serial1

Adafruit_Fingerprint finger = Adafruit_Fingerprint(&mySerial);

uint8_t getFingerprintEnroll();
void printHex(int num, int precision);

void setup()
{
    Serial.begin(9600);
    Serial.println("Adafruit Fingerprint sensor enrollment");

    // set the data rate for the sensor serial port
    finger.begin(57600);

    if (finger.verifyPassword()) {
        Serial.println("Found fingerprint sensor!");
    } else {
        Serial.println("Did not find fingerprint sensor :(");
        while (1);
    }
}

void loop() {
    ;
}

void serialEvent() {
    if (Serial.available()) {
        char c = Serial.read();
        if (c == '1')
            getFingerprintEnroll();
    }
}

uint8_t getFingerprintEnroll() {

    int p = -1;
    Serial.print("Waiting for valid finger to enroll..");
    while (p != FINGERPRINT_OK) {
        p = finger.getImage();
        switch (p) {
            case FINGERPRINT_OK:
                Serial.println("Image taken");
                break;
            case FINGERPRINT_NOFINGER:
                Serial.println(".");
                break;
            case FINGERPRINT_PACKETRECIEVEERR:
                Serial.println("Communication error");
                break;
            case FINGERPRINT_IMAGEFAIL:
                Serial.println("Imaging error");
                break;
            default:
                Serial.println("Unknown error");
                break;
        }
    }

    // OK success!

    p = finger.image2Tz(1);
    switch (p) {
        case FINGERPRINT_OK:
            Serial.println("Image converted");
            break;
        case FINGERPRINT_IMAGEMESS:
            Serial.println("Image too messy");
            return p;
        case FINGERPRINT_PACKETRECIEVEERR:
            Serial.println("Communication error");
            return p;
        case FINGERPRINT_FEATUREFAIL:
            Serial.println("Could not find fingerprint features");
            return p;
        case FINGERPRINT_INVALIDIMAGE:
            Serial.println("Could not find fingerprint features");
            return p;
        default:
            Serial.println("Unknown error");
            return p;
    }

    Serial.println("Remove finger");
    delay(500);
    p = 0;
    while (p != FINGERPRINT_NOFINGER) {
        p = finger.getImage();
    }
    Serial.println("...");
    p = -1;
    Serial.println("Place same finger again");
    while (p != FINGERPRINT_OK) {
        p = finger.getImage();
        switch (p) {
            case FINGERPRINT_OK:
                Serial.println("Image taken");
                break;
            case FINGERPRINT_NOFINGER:
                Serial.print(".");
                break;
            case FINGERPRINT_PACKETRECIEVEERR:
                Serial.println("Communication error");
                break;
            case FINGERPRINT_IMAGEFAIL:
                Serial.println("Imaging error");
                break;
            default:
                Serial.println("Unknown error");
                break;
        }
    }

    // OK success!

    p = finger.image2Tz(2);
    switch (p) {
        case FINGERPRINT_OK:
            Serial.println("Image converted");
            break;
        case FINGERPRINT_IMAGEMESS:
            Serial.println("Image too messy");
            return p;
        case FINGERPRINT_PACKETRECIEVEERR:
            Serial.println("Communication error");
            return p;
        case FINGERPRINT_FEATUREFAIL:
            Serial.println("Could not find fingerprint features");
            return p;
        case FINGERPRINT_INVALIDIMAGE:
            Serial.println("Could not find fingerprint features");
            return p;
        default:
            Serial.println("Unknown error");
            return p;
    }

    // OK converted!
    Serial.print("Creating model....");

    p = finger.createModel();
    if (p == FINGERPRINT_OK) {
        Serial.println("Prints matched!");
    } else if (p == FINGERPRINT_PACKETRECIEVEERR) {
        Serial.println("Communication error");
        return p;
    } else if (p == FINGERPRINT_ENROLLMISMATCH) {
        Serial.println("Fingerprints did not match");
        return p;
    } else {
        Serial.println("Unknown error");
        return p;
    }

    p = finger.getModel();

    if (p == FINGERPRINT_OK) {
        Serial.println("transferrring...");
    } else if (p == FINGERPRINT_PACKETRECIEVEERR) {
        Serial.println("Communication error");
        return p;
    } else if (p == FINGERPRINT_UPLOADFEATUREFAIL) {
        Serial.println("Transmission error");
        return p;
    } else {
        Serial.println("Unknown error");
        return p;
    }

    // one data packet is 267 bytes
    // in one data packet, 11 bytes are 'usesless' :D
    uint8_t bytesReceived[534]; // 2 data packets
    memset(bytesReceived, 0xff, 534);

    uint32_t starttime = millis();
    int i = 0;
    while (i < 534 && (millis() - starttime) < 20000) {
        if (mySerial.available()) {
            bytesReceived[i++] = mySerial.read();
        }
    }
    Serial.print(i); Serial.println(" bytes read.");
    Serial.println("decoding packet...");

    uint8_t fingerTemplate[512]; // the real template
    memset(fingerTemplate, 0xff, 512);

    // filtering only the data packets
    int uindx = 9, index = 0;
    while (index < 534) {
        while (index < uindx) ++index;
        uindx += 256;
        while (index < uindx) {
            fingerTemplate[index++] = bytesReceived[index];
        }
        uindx += 2;
        while (index < uindx) ++index;
        uindx = index + 9;
    }
    for (int i = 0; i < 512; ++i) {
        //Serial.print("0x");
        printHex(fingerTemplate[i], 2);
        //Serial.print(", ");
    }
    Serial.println("done.");
}

void printHex(int num, int precision) {
    char tmp[16];
    char format[128];

    sprintf(format, "%%.%dX", precision);

    sprintf(tmp, format, num);
    Serial.print(tmp);
}

`

drichardcarl, 2017-11-02:
@Obaid-ur-Rehman the sample code is here :D
[https://pastecode.xyz/view/425c6b1f](url)

Try-Parser, 2017-11-22:
i used the upChar for getting the template and i used the downChar to load the template but its sad to say that it will automatically detect the last enrolled in the sensor, the matcher which is  in the internal of the sensor will match the last fprint value and the new capture one, even i didn't save the template on its memory, i dont know whats going on inside the sensor anymore if someone has done this before and made it success can some one share some of the info please.

Try-Parser, 2017-11-23:
I think i found it how to heheh thanks guys..

Try-Parser, 2017-11-23:
I am now able to get then finger print template and store in a db using base64 encoding and load it to the sensor for match.  it should be 688 bytes with the headers

Obaid-ur-Rehman, 2017-11-26:
@drichardcarl Thank you.
Unfortunately the link does not works, could be broken. Can you please share an alternative link?

Obaid-ur-Rehman, 2017-11-26:
@Try-Parser Can you please share the source code. Thank you

Try-Parser, 2017-11-27:
@Obaid-ur-Rehman you can pm me or ping me in email. ferox.dragon@gmail.com

ladyada, 2017-11-28:
@Try-Parser the code has been added to the show_fingerprint_templates example!

vilipwong, 2017-12-14:
Is there a way to upload template into the sensor for matching ?

Try-Parser, 2017-12-15:
Yes it has.

Try-Parser, 2017-12-15:
function name is upChar

sqwerty13, 2017-12-15:
Yes, there is upChar in datasheets, but there isn't upChar function in the library...

ricarduscat, 2017-12-16:
Yes, I would like to have this function implemented, I tried to fix the library but doesn't work... Anyone can help me? I just opened a new issue

one-player, 2017-12-18:
How to synchronize prints between two modules. Tell me please.
For example, if I use two modules and one will fail. How do I make a module dump in order to unload it into a new module?

ricarduscat, 2017-12-18:
I'm doing this function, to put external templates into the flash memory of the sensor. This is the function:

`uint8_t Adafruit_Fingerprint::downloadModel(uint16_t id, uint8_t packet_1[], uint8_t packet_2[], uint8_t packet_3[], uint8_t packet_4[], uint8_t slot) {
  uint8_t packet[] = {FINGERPRINT_DOWNLOAD, slot};
  writePacket(theAddress, FINGERPRINT_COMMANDPACKET, sizeof(packet)+2, packet);
  uint8_t len = getReply(recvPacket);
  uint8_t x;
  if ((len != 1) && (recvPacket[0] != FINGERPRINT_ACKPACKET))
   return -1;
  if (recvPacket[1] == 0x00) {
    writePacket(theAddress, 0x02, 66, packet_1);
    writePacket(theAddress, 0x02, 66, packet_2);
    writePacket(theAddress, 0x02, 66, packet_3);
    writePacket(theAddress, 0x08, 66, packet_4);
    return recvPacket[1];
  } else {
    return -1;
  }
}`

This works. You have to call this function in the arduino code and it saves the fingerprint to buffer1, later, you have to call the SaveModel function to save the template with the ID. But there is a problem. There is no control ACK packets for the datapackets you send, so you can't control if the entire template is reaching the sensor. The decumentations says nothing about it... so... really?? No ACK paquets for every datapacket?? The down Char functon it's totally useless without this control... You will never be able to put an entire template to buffer 1

marianodato, 2018-03-10:
Anyone found a solution to upload a template to the sensor?

chhass, 2018-03-12:
@drichardcarl did you solve this problem, matching fingerprint templates data?
I mean comparing data directly, outside module?

mgonzaga1990, 2018-03-13:
Hi guys,
Is there a way where I can get the ff:

1. get the equivalent hash or byte value of my FP and send it over to the backend via Serial for saving to DB.
2.get the earlier FP template sent via Serial then use it for the Sensor to match against the current FP which i just put into the sensor?

My goal is to save the FP to the DB and later on retrieve via Arduino UNO and FP sensor to compare against different FP until matched.

Thanks

chhass, 2018-03-13:
@mgonzaga1990,

you can do that, but, perhaps serial data rate, can take a lot of time if you need to search several FP's.

With 115200bps, you will be able to send 28 FP's/second, that's too slow!

We need a algorith to match FP template/char file.

mgonzaga1990, 2018-03-14:
@chhass 
I'm working on a project where the FP from the sensor should be saved in the database.
I have a frontend (VB.NET) and backend (Java) and database connected to the backend.

the FP sensor is connected to the frontend via serial. While VB.net is connected to Java via RestAPI.

so initially, the user would sign in and register his FP (that's why i need the FP template and save it in the DB). then later on once the user has login using his FP. The FP stored in the DB will be used to compare against the FP template that he use to login.

joseluissegura, 2018-05-23:
I send to the fp-module the following package  0xEF01 0xFFFFFFFF 0x08 0x01 sum
and it returns 0xEF01 0xFFFFFFFF 0x07 0x0003 0x00 0x000A (ack package saying "ready to transfer the following data packet")
I wait 5 secs and starts reading the serial port
but it only receives 0xEF01 0xFFFFFFFF 0x02 0x0042 (and 54 more bytes) it never sends more than that,
no matter how much time I wait for the rest of the bytes.

I have 2 fp scanners and they both return the same

ps: I have the sysparam of the package size set to 64 
I already tried 32, 128 and 256

dheeptuck, 2018-06-29:
I am also looking for a way to authenticate the captured fingerprint outside the FP sensor.

kamsri81, 2018-07-03:
Hi dheeptuck,
                Did you have any luck with the DownChar() command? Whenever i write the data (Same data which i read using UpChar() command) back to the sensor it gives error( 0x01h: error when receiving data package).I have contacted the Rhydolabz support team but they have still not got back on this.

beerni, 2018-08-18:
Any news on this?

sand007man, 2018-10-01:
> I notice I could extract a fingerprint's template from the library. I wonder whether there is anyway to upload the template to my server where I can compute the matching score.
>
> (As the FP sensor [...]

i am able to enroll ,search fingerprint but when i load show_fingperprint _template code it is showing some "??" values in the serial monitor . even small help will be greatly appreciated . I am using this fingerprint with node_mcu esp 12e...i feel its something related to memory buffer !!

sand007man, 2018-10-03:
> I am now able to get then finger print template and store in a db using base64 encoding and load it to the sensor for match. it should be 556 + 12 = 568 bytes with the headers

hi can you tell how did you upload template into db ....i am getting 768 byte of template data from the fingerprint sensor i am using !! in which format i have convert the template so that it can be stored in mysql db..right now all template is store in buffer.

qwerty13, 2018-11-04:
Any Success for make use of DownChar?

janioanselmo, 2018-12-28:
Hi, @dheeptuck and everyone else interested in validating the "template" externally! I have been developing a solution for access control for a long time. However, a large majority of readers (financially viable) can not do external validation (servers - LINUX/Windows/MAC). What I have evaluated and accomplished is the work of a genius, named: Robert Važan - https://github.com/robertvazan/sourceafis-java
Have a look, incredible!

Try-Parser, 2019-01-02:
I written a little bit script using python you can check here 
https://github.com/Try-Parser/FsSoftware/blob/master/GTSensor.py
i am really sorry i cannot make a PR i quite busy. @ladyada you can convert it to c++

hanifizzudinrahman, 2019-02-21:
> I'm doing this function, to put external templates into the flash memory of the sensor. This is the function:
>
> `uint8_t Adafruit_Fingerprint::downloadModel(uint16_t id, uint8_t packet_1[], uint8_t [...]

What "recvPacket" is? The last packet? or what?

hanifizzudinrahman, 2019-03-29:
**Alhamdulillah**, i have finished our Project for Research and Development for Fingerprint Module (with ZFM20 or DY50/FPM10A)
This is include file C for Arduino and surely the libraries
Enroll + Match + Upload Template + Show Template + Delete & Empty + Show Image Template + SFG Demo

So, i uploaded in github, you can visit it in
https://github.com/hanifizzudinrahman/Module-Fingerprint-DY50-FPM10A

hanifizzudinrahman, 2019-04-01:
**Fingerprint Module**
[image]
**Schematic**
[image]

abhishekgautam101, 2019-04-09:
### please refer to my code
[https://github.com/abhishekgautam101/r30x-fingerprint-sensor-library-for-arduino](url)

### I am using the following code to push buffer 1 template(which I got by using upChar) from Arduino to fingerprint sensor
`void match(){
   while (!Serial.available());
   String str;
   for(int i=0; i< 147; i++){             //reading string from serial
    if(Serial.available()){
      str += Serial.read();
    }
   }
   
  byte byte_arr[63];
  int arr_element_index = 0;
  int startIndex = 0;
  int endIndex = 0;
  for(int i=0 ; i<str.length() ; i++){          //converting string to hex byte array
    if(str.charAt(i) == ','){
      endIndex = i;
      String str_part =  str.substring(startIndex, endIndex);
      char charBuf[2];
      str_part.toCharArray(charBuf, 2);
      byte_arr[arr_element_index] = (byte) strtoul(charBuf, NULL, 16);
      startIndex = endIndex + 1;
      arr_element_index++;
    }
  }

   r30x_fl.begin(115200);        //sending data to fingerprint module
   delay(1000);

   if(r30x_fl.verifyPassFunc()){
    if(r30x_fl.callGetImage()){
      if(r30x_fl.img2buffer2Func()){
        //feeding previous data of user to fingerprint to match
          if(r30x_fl.downCharBuffer1Func(byte_arr)){
            if(r30x_fl.matchBuffersFunc()){
              delay(1000);
               Serial.print("~Fingers mached!$");
            }else{
              delay(1000);
              Serial.print("~Fingers did not match! Try again...$");
            }
          }else{
            delay(1000);
            Serial.print("~Unknown error! Try again...$");
          }
      }else{
        delay(1000);
        Serial.print("~Finger not Captured properly! Try again...$");
      }
    }else{
      delay(1000);
      Serial.print("~Time Out! Try again...$");
    }
  }else{
    delay(1000);
    Serial.print("~Incorrect credential! Try again...$");
  }
}
`
the template buffer which I am sending is 
`EF,1,FF,FF,FF,FF,2,0,82,3,1,4D,1C,86,0,80,6,80,6,0,6,0,6,0,6,0,2,80,2,80,2,80,6,C0,6,C0,6,C0,6,E0,1E,F0,3E,FF,FE,FF,FE,FF,FE,0,0,0,0,0,0,0,0,18,18,BD,27,EE,8F,`
as a string , then decoding it in above function.

Please tell me what I am doing wrong!

Arpit-exe, 2019-05-08:
I have modified and combined adafruit's enroll code with show fingerprint template to generate a workable code which scans a finger gets the image and converts it to a template and upload it to the serial monitor. Successfully

bekelemulugeta, 2019-05-25:
#Arpit-exe#could you give me your code please?

hanifizzudinrahman, 2019-06-01:
> **Alhamdulillah**, i have finished our Project for Research and Development for Fingerprint Module (with ZFM20 or DY50/FPM10A)
> This is include file C for Arduino and surely the libraries
> Enroll + [...]

this code can help you. dont forget to change the library, because it's corresponding with the code

bekelemulugeta, 2019-06-04:
thank you very much

Abrie1995, 2019-06-11:
@bekelemulugeta did you manage to match a template to the live finger on the fingerprint sensor? or did anyone got this to work?

chadianscot, 2019-06-18:
I've just started playing with extracting templates from a JM-101B which seems to be a ZFM-20 essentially. My problem is that when I send the UPCHAR command to extract the template, I get 846 bytes back (ACK packet and headers included). Based on hanifizzudinrahman's project, I should be expecting 554 bytes. Has anyone else seen this difference?

hanifizzudinrahman, 2019-06-18:
> I've just started playing with extracting templates from a JM-101B which seems to be a ZFM-20 essentially. My problem is that when I send the UPCHAR command to extract the template, I get 846 bytes [...]

UPCHAR is upload template to upper computer (0x08)
When i use ZFM20, i get 4 lines that in 1 line includes 139 bytes data
- 9 bytes data
   --> 2 bytes Header ( EF 01)
   --> 4 bytes Address (FF FF FF FF)
   --> 1 byte Identifier (02)
   --> 2 bytes Length (00 82)
                 # "00 82" means in biner "130" the length of data template + check sum
- 128 bytes data
   --> **This is the important data**, because this is **the data template of our fingerprint**
                # "03 03 ..... 00", this is the part of my template fingerprint
- 2 bytes data
   --> This is the data of Check sum (12 DF)
               # The Sum of "Identifier" , "Length" and "Data Template"
==> So, in 1 line i get (9 + 128 + 2) bytes = 139 bytes
and in 4 lines, i get (139 x 4) bytes = 556 bytes

Honestly, i limit the UPCHAR command only in 4 lines. You can change it in the arduino code "ZFM20_ShowTemplate.ino"
Why i limit it? because **the data template fingerprint** after line 4 is ZERO (00 00 ..... 00). So, this is the data default of template fingerprint when there is no more data from module fingerprint.
-------------------------------------------------------------------------------------------------------------------------------------------
And from chadianscot's problem, make sure that the 846 bytes data, there is no zero data
Even though that 846 bytes data is right data, you can still upload your **data template fingerprint** from upper computer to module (0x09) with little modification in the library and header ".uploadModelPercobaan(packet2,....,packet5)"

i'm so glad to help someone problem. Ask me again when you have any problem... :D

danilpan, 2019-06-26:
Guys, how can i extract template from this sensor?

hanifizzudinrahman, 2019-06-27:
To extract template fingerprint from module fingerprint to mikrokontroler
1. from mikrokontroler, you have to send command "download 0x08" to module fingerprint. you can look in datasheet of your module fingerprint
2. after that, module fingerprint will reply your command
3. next, module fingerprint will send you a package of **template fingerprint**
-- Module Fingerprint DY50 (6 package)
[image]
-- Module Fingerprint ZFM20 (4 package)

4. After that, You must filter it to pull out only the **Template Fingerprint**. you can visit my github project

danilpan, 2019-06-28:
Thank you for your answer hanifizzudinrahman. Can I take Template without comparing with other fingerprints? How much time it will take? Could you help with it?

Anisank25, 2019-07-29:
> 
> 
> I am now able to get then finger print template and store in a db using base64 encoding and load it to the sensor for match. it should be 556 + 12 = 568 bytes with the headers

Can you explain or attach code that you used to store it?

kejuqeju, 2019-09-13:
@hanifizuddin
i am experiencing same thing as you did. are you sure that the template data in the 5th packet and following packets always 0 ? have you try with different fingerprints?
second question: when i send command upchar (sends template data to CharBuff), do i have to send all the packet data or just the first 4 packets ?

hanifizzudinrahman, 2019-09-14:
> Thank you for your answer hanifizzudinrahman. Can I take Template without comparing with other fingerprints? How much time it will take? Could you help with it?

yes you can take the template without comparing it. but, about the time, i forget. i have calculate it, but i forget. i think, the time is less than 1 second

hanifizzudinrahman, 2019-09-14:
> @hanifizuddin
> i am experiencing same thing as you did. are you sure that the template data in the 5th packet and following packets always 0 ? have you try with different fingerprints?
> second [...]

yes, i have try it with different fingerprints. It appears zero because the module only save a packet of template fingerprint according on datasheet. you can read using "read parameter"
in second question, refers on my first answer. the charBuff on module is limited, according on datasheet. So, only first 4 packets that will sent

kejuqeju, 2019-09-15:
i just checked and tried some fingerprints. it is true that for some fingerprints, they only need 4 data packets. the rest 8 packets are zeros. but if i tried larger fingers, like thumbs, i saw that it needs 6 data packets.

danwillm, 2019-09-28:
Was there anything happening on this? I to want to get something to compare templates working as well. Thanks!

yvesniyo, 2019-11-27:
> UPCHAR is upload template to upper computer (0x08)
> When i use ZFM20, i get 4 lines that in 1 line includes 139 bytes data
>
> * 9 bytes data
>   --> 2 bytes Header ( EF 01)
>   --> 4 bytes Address (FF FF [...]

I have tried so many times by using this function "uploadModelPercobaan()" with my saved packet2 to packet5 and but none has ever matched, please is it possible to get it compared in fingerprint device or not going to happen so that I can start to look for how I can extract image and then compare them with another program

rehamouv303, 2019-12-07:
> To extract template fingerprint from module fingerprint to mikrokontroler
>
> 1. from mikrokontroler, you have to send command "download 0x08" to module fingerprint. you can look in datasheet of your [...]

How can I extract template from fingerprint module R307 !!  Using uno... can you give me code for it

teezzan, 2019-12-07:
@hanifizzudinrahman , Thank you very much. It works very well. 
Jazakumullahu khyran.

ayudavisual, 2019-12-09:
Hello, I'm using dy50 sensor but I can't save the 7 packets,  someone knows how do this? I only can save  packed2 to packed5

[51 more replies]