adafruit/Adafruit-Fingerprint-Sensor-Library #16: unknown error

ajaykrishna97, 2016-09-20:
on enrolling process ,after scanning finger print ,an unknown error occurs,plz help me to rectify it,i will be happy if i get suggestions on some common unknown error..

tborja, 2016-10-03:
I am getting the same error. I am using a Bluno Nano and MySerial on 2 and 3. The Unknown Error happens on the call to this method Adafruit_Fingerprint::createModel.

sumitswain, 2016-10-03:
I am getting the same error . from fingerprint:createModel

Blacksincan, 2016-10-03:
please fix this .. i have same problem .. unknown erorr

rohithkrishnan, 2016-10-03:
Tried enrolling with SFG software and it works fine. But when I try enrolling with Arduino it throws the error (from the function call of CreateModel). Somehow the value of p is being set to 0xB8 whereas it should have been 0x00.

nasreddine27, 2016-10-03:
try my code
`#include <Adafruit_Fingerprint.h>
#include <SoftwareSerial.h>
uint8_t id;
int getFingerprintEnroll();
SoftwareSerial mySerial(2, 3);
Adafruit_Fingerprint finger = Adafruit_Fingerprint(&mySerial);
void setup()  
{
  Serial.begin(9600);
  finger.begin(57600);
}
uint8_t readnumber(void) {
  uint8_t num = 0;
  while (Serial.available()==0);
  num = Serial.parseInt();
  return num;
}
void loop()
{
  Serial.println("Pret pour enroller une empreinte! Tapez l'ID s'il vous plait...");
  id = readnumber();
  Serial.print("Enrollement de l'ID #");
  Serial.println(id);
  
  while (!  getFingerprintEnroll() );
}
int getFingerprintEnroll() {
  int p = -1;
  Serial.print("En attente d'un doigt valide pour l'enroller en tant que ID #"); Serial.println(id);
  while (p != FINGERPRINT_OK) {
    p = finger.getImage();
    switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image prise");
      break;
    case FINGERPRINT_NOFINGER:
      Serial.println(".");
      break;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Erreur de communication");
      break;
    case FINGERPRINT_IMAGEFAIL:
      Serial.println("Erreur d'image");
      break;
    default:
      Serial.println("Erreur inconnue");
      break;
    }
  }
  p = finger.image2Tz(1);
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("L'image est convertie");
      break;
    case FINGERPRINT_IMAGEMESS:
      Serial.println("Image trés salissante");
      return p;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Erreur de communication");
      return p;
    case FINGERPRINT_FEATUREFAIL:
      Serial.println("Impossible de trouver des caractéristiques de l'empreinte digitale");
      return p;
    case FINGERPRINT_INVALIDIMAGE:
      Serial.println("Impossible de trouver des caractéristiques de l'empreinte digitale");
      return p;
    default:
      Serial.println("Erreur inconnue");
      return p;
  }
  Serial.println("Retirez votre doigt s'il vous plait");
  delay(2000);
  p = 0;
  while (p != FINGERPRINT_NOFINGER) {
    p = finger.getImage();
  }
  Serial.print("ID "); Serial.println(id);
  p = -1;
  Serial.println("Placez le même doigt à nouveau");
  while (p != FINGERPRINT_OK) {
    p = finger.getImage();
    switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image prise");
      break;
    case FINGERPRINT_NOFINGER:
      Serial.print(".");
      break;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Erreur de communication");
      break;
    case FINGERPRINT_IMAGEFAIL:
      Serial.println("Erreur d'image");
      break;
    default:
      Serial.println("Erreur inconnue");
      break;
    }
  }
  p = finger.image2Tz(2);
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("L'image est convertie");
      break;
    case FINGERPRINT_IMAGEMESS:
      Serial.println("Image trés salissante");
      return p;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Erreur de communication");
      return p;
    case FINGERPRINT_FEATUREFAIL:
      Serial.println("Impossible de trouver des caractéristiques de l'empreinte digitale");
      return p;
    case FINGERPRINT_INVALIDIMAGE:
      Serial.println("Impossible de trouver des caractéristiques de l'empreinte digitale");
      return p;
    default:
      Serial.println("Erreur inconnue");
      return p;
  }
  
  // OK converted!
  Serial.print("Création d'un modèle pour #");  Serial.println(id);
  
  p = finger.createModel();
  if (p == FINGERPRINT_OK) {
    Serial.println("Empreintes matched!");
  } else if (p == FINGERPRINT_PACKETRECIEVEERR) {
    Serial.println("Erreur de communication");
    return p;
  } else if (p == FINGERPRINT_ENROLLMISMATCH) {
    Serial.println("Les empreintes se different");
    return p;
  } else {
    Serial.println("Erreur inconnue");
    return p;
  }   
  
  Serial.print("ID "); Serial.println(id);
  p = finger.storeModel(id);
  if (p == FINGERPRINT_OK) {
    Serial.println("Enregistrer!");
  } else if (p == FINGERPRINT_PACKETRECIEVEERR) {
    Serial.println("Erreur de communication");
    return p;
  } else if (p == FINGERPRINT_BADLOCATION) {
    return p;
  } else if (p == FINGERPRINT_FLASHERR) {
    return p;
  } else {
    Serial.println("Erreur inconnue");
    return p;
  }   
}`

nasreddine27, 2016-10-03:
[enroll.txt](https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/files/506941/enroll.txt)
try my code

Blacksincan, 2016-10-04:
stil same unknown erorr... im using arduino uno r3 .. pin 2,3 
[image]

lspoplove, 2016-10-09:
same error

tborja, 2016-10-09:
Same error using Bluno Nano pin 2,3

bragajames2, 2016-10-09:
same problem. i am using Arduino Uno, pin 2,3 .. errors still occur specially on case FINGERPRINT_FEATUREFAIL

rohithkrishnan, 2016-10-09:
https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/pull/15

Tested and working.

Grunewald17, 2016-12-17:
use Arduino IDE 1.6.7 ... then its working fine. 1.6.13 has an array issue.

GrexAut, 2017-01-01:
This works with latest Arduino 1.8.0: https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/pull/18

rengelking, 2017-01-02:
This is still causing some errors; I can get it to work with 1.6.7, but in both 1.6.13 and 1.8.0, when just adding another finger.getImage(); command in a subroutine that gets compiled (but never called), it will cause an error... very very strange.  When the error occurs, I am getting the B8 returned from the finger.image2Tz routine for some reason.  I have done a memory check, and it has nothing to do with memory being low; whenever another call to the finger.getImage() is added, it will cause this error, even if I put that in separate subroutine and call it from both of them.

rengelking, 2017-01-02:
I did some tests and just adding the extra finger.getImage() statement, even if it never gets called, causes very strange array errors.  This is what I added to the image2Tz function to test a bunch of things:
```
  uint8_t Adafruit_Fingerprint::image2Tz(uint8_t slot) {
  uint8_t packet[] = {FINGERPRINT_IMAGE2TZ, slot};
  byte packet2[2];
  packet2[0]=FINGERPRINT_IMAGE2TZ;
  packet2[1]=slot;
//  Serial.print("&packet[0]:"); Serial.println(&packet[0],HEX);
  packet[0]=2;
  uint8_t bob=FINGERPRINT_IMAGE2TZ;
  
  Serial.print("\npacket[0]:"); Serial.println(packet[0],HEX);
  Serial.print("bob:"); Serial.println(bob,HEX);
  Serial.print("FINGERPRINT_IMAGE2TZ:"); Serial.println(FINGERPRINT_IMAGE2TZ,HEX);
  for(int i=0;i<4;i++) {
    Serial.print("i:"); Serial.print(i); Serial.print("\t"); Serial.println(packet[i],HEX);
  }
  for(int i=0;i<4;i++) {
    Serial.print("2 i:"); Serial.print(i); Serial.print("\t"); Serial.println(packet2[i],HEX);
  }
  Serial.print("sizeof(packet)+2:"); Serial.println(sizeof(packet)+2);
  writePacket(theAddress, FINGERPRINT_COMMANDPACKET, sizeof(packet)+2, packet);
  uint8_t len = getReply(packet);
  
  if ((len != 1) && (packet[0] != FINGERPRINT_ACKPACKET))
   return -1;
  return packet[1];
}
```

Without the extra finger.getImage() statement, I get the following results:
```
packet[0]:2
bob:2
FINGERPRINT_IMAGE2TZ:2
i:0	2
i:1	1
i:2	EE
i:3	0
2 i:0	2
2 i:1	1
2 i:2	0
2 i:3	B
sizeof(packet)+2:4
```
 
With the extra finger.getimage() statement (never called), I get the following bizarre results where none of the packet or packet2 array variables can hold a correct value:
```
packet[0]:B8
bob:2
FINGERPRINT_IMAGE2TZ:2
i:0	B8
i:1	B8
i:2	B8
i:3	B8
2 i:0	7
2 i:1	0
2 i:2	0
2 i:3	0
sizeof(packet)+2:4
```

Clearly there is a bug in the compilers 1.6.13 and 1.8.0 that still exists.

rengelking, 2017-01-03:
I rewrote the library as a work around to the 1.8.0 problem, because it definitely WAS NOT FIXED.  There were still problems as I mentioned above, based on the sketch.

To correct the problem I added a buffer to the class definition in the .h file (packetBuffer[64]):
```
class Adafruit_Fingerprint {
 public:
#ifdef __AVR__
  Adafruit_Fingerprint(SoftwareSerial *);
#endif
  Adafruit_Fingerprint(HardwareSerial *);
  uint8_t packetBuffer[64];

```
then updated all the class functions to point to that buffer instead of creating one in each function:
```
/*************************************************** 
  This is a library for our optical Fingerprint sensor

  Designed specifically to work with the Adafruit Fingerprint sensor 
  ----> http://www.adafruit.com/products/751

  These displays use TTL Serial to communicate, 2 pins are required to 
  interface
  Adafruit invests time and resources providing this open source code, 
  please support Adafruit and open-source hardware by purchasing 
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.  
  BSD license, all text above must be included in any redistribution
 ****************************************************/

#include "Fingerprint.h"

#ifdef __AVR__
    #include <util/delay.h>
    #include <SoftwareSerial.h>
#endif

#ifdef __AVR__
Adafruit_Fingerprint::Adafruit_Fingerprint(SoftwareSerial *ss) {
  thePassword = 0;
  theAddress = 0xFFFFFFFF;

  hwSerial = NULL;
  swSerial = ss;
  mySerial = swSerial;
}
#endif

Adafruit_Fingerprint::Adafruit_Fingerprint(HardwareSerial *ss) {
  thePassword = 0;
  theAddress = 0xFFFFFFFF;

#ifdef __AVR__
  swSerial = NULL;
#endif
  hwSerial = ss;
  mySerial = hwSerial;
}

void Adafruit_Fingerprint::begin(uint16_t baudrate) {
  delay(1000);  // one second delay to let the sensor 'boot up'

  if (hwSerial) hwSerial->begin(baudrate);
#ifdef __AVR__
  if (swSerial) swSerial->begin(baudrate);
#endif
}

boolean Adafruit_Fingerprint::verifyPassword(void) {
[... 258 lines cut ...]
    if (idx >= 9) {
      if ((reply[0] != (FINGERPRINT_STARTCODE >> 8)) ||
          (reply[1] != (FINGERPRINT_STARTCODE & 0xFF)))
          return FINGERPRINT_BADPACKET;
      uint8_t packettype = reply[6];
      //Serial.print("Packet type"); Serial.println(packettype);
      uint16_t len = reply[7];
      len <<= 8;
      len |= reply[8];
      len -= 2;
      //Serial.print("Packet len"); Serial.println(len);
      if (idx <= (len+10)) continue;
      packet[0] = packettype;      
      for (uint8_t i=0; i<len; i++) {
        packet[1+i] = reply[9+i];
      }
#ifdef FINGERPRINT_DEBUG
      Serial.println();
#endif
      return len;
    }
  }
}

```

RajaViki, 2017-01-30:
The people who all have "Unknown Error" issue try using Arduino Complier of lower version like 1.0.6 and below ,
that will solve your error as the latest Arduino Compliers are not stable please use lower versions

rengelking, 2017-01-30:
The latest compilers are stable, they just have errors when compiling this type of dynamically declared class array. Switching back to lower versions of the compiler is not an option for people that have many projects that require the newest compiler features Rewriting the library slightly, like I have done above, is a much better solution and is entirely stable.

RajaViki, 2017-01-31:
Hi RengelKing
Before this I have another strange issue , 10 days back my fingerprint module was working properly and executing ,suddenly it is not get through the password verification and shows "Fingerprint sensor not found :(" .I was running the same code till the day before but it is not working now, I have tried replacing new library ,new Arduino board and even new PC but it does seem to be solving .Using thesoftware enrollment "SFG Demo" it is getting throu the password and works.
Can you help me out ?

therealmcgrath, 2017-01-31:
Hello all,

I am having similar problem getting to the enrol stage and "unknown error" shows up for every finger.

I have macbook pro and using Adruino 1.8.1.

Ive read all problems and I have tried everything from new code to downgrading Aduino 1.6.6.

And still not working ?????

RajaViki, 2017-02-01:
try 
Arduino version 1.0.6 that will work for sure

therealmcgrath, 2017-02-01:
I tried the code giving in tutriol, i also tried 1.0.6 and it does not work.

What I'm i doing wrong?

Could this have anything to do with a MacBook Pro.
[image]

RajaViki, 2017-02-02:
Error shows that ,it had not recognised the library file 
So it might know the location for llibrary file , go to File->Preference and add the location where u have extracted your library file or add the library thru sketch->Import library -> Add Library -> and add that Adafruit_Fingerprint_Master.zip

Kiwikisser, 2017-02-03:
Got it to work with Arduino 1.6.13, no more unknown errors

adwidianjaya, 2017-03-03:
#15 also works for me with Arduino 1.6.13.

dello14, 2017-06-19:
We're having the same problem with the FPS.

We're using a Wido board and an LCD shield and FPS for our project.

We've tried to use the FPS on the wido board but after entering 1 and putting my thumbmark it won't respond.

We've try to change the serial port into 8, 9 since it's a leonardo.

But the thing is we've also tried to use the arduino and the FPS work but after putting our thumbmark it said "Unknown Error"

We don't know if it's the wire or the codes but the codes we use is the example code of the Adafruit.

We prepare to use the Wido board if happen that the FPS work on the Wido board we will combine the wido and the LCD shield and use the FPS.

Hope someone can resolve this issue.

therealmcgrath, 2017-07-06:
Please Help Unknown error

suryapraneeta, 2017-10-12:
For this unknown error we need to just rewrite the header file and .cpp file. the data reading from the sensor has to be stored in buffer.
--------------------------------------------------------------------------------
Header file has to be modified as 
/*************************************************** 
  This is a library for our optical Fingerprint sensor

  Designed specifically to work with the Adafruit Fingerprint sensor
  ----> http://www.adafruit.com/products/751

  These displays use TTL Serial to communicate, 2 pins are required to 
  interface
  Adafruit invests time and resources providing this open source code, 
  please support Adafruit and open-source hardware by purchasing 
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.  
  BSD license, all text above must be included in any redistribution
 ****************************************************/

#if (ARDUINO >= 100)
 #include "Arduino.h"
 #include <SoftwareSerial.h>
#else
 #include "WProgram.h"
 #include <NewSoftSerial.h>
#endif

#define FINGERPRINT_OK 0x00
#define FINGERPRINT_PACKETRECIEVEERR 0x01
#define FINGERPRINT_NOFINGER 0x02
#define FINGERPRINT_IMAGEFAIL 0x03
#define FINGERPRINT_IMAGEMESS 0x06
#define FINGERPRINT_FEATUREFAIL 0x07
#define FINGERPRINT_NOMATCH 0x08
#define FINGERPRINT_NOTFOUND 0x09
#define FINGERPRINT_ENROLLMISMATCH 0x0A
#define FINGERPRINT_BADLOCATION 0x0B
#define FINGERPRINT_DBRANGEFAIL 0x0C
#define FINGERPRINT_UPLOADFEATUREFAIL 0x0D
#define FINGERPRINT_PACKETRESPONSEFAIL 0x0E
#define FINGERPRINT_UPLOADFAIL 0x0F
#define FINGERPRINT_DELETEFAIL 0x10
#define FINGERPRINT_DBCLEARFAIL 0x11
#define FINGERPRINT_PASSFAIL 0x13
#define FINGERPRINT_INVALIDIMAGE 0x15
#define FINGERPRINT_FLASHERR 0x18
#define FINGERPRINT_INVALIDREG 0x1A
#define FINGERPRINT_ADDRCODE 0x20
#define FINGERPRINT_PASSVERIFY 0x21

#define FINGERPRINT_STARTCODE 0xEF01

#define FINGERPRINT_COMMANDPACKET 0x1
#define FINGERPRINT_DATAPACKET 0x2
#define FINGERPRINT_ACKPACKET 0x7
#define FINGERPRINT_ENDDATAPACKET 0x8

#define FINGERPRINT_TIMEOUT 0xFF
#define FINGERPRINT_BADPACKET 0xFE

#define FINGERPRINT_GETIMAGE 0x01
#define FINGERPRINT_IMAGE2TZ 0x02
#define FINGERPRINT_REGMODEL 0x05
#define FINGERPRINT_STORE 0x06
#define FINGERPRINT_LOAD 0x07
#define FINGERPRINT_UPLOAD 0x08
#define FINGERPRINT_DELETE 0x0C
#define FINGERPRINT_EMPTY 0x0D
#define FINGERPRINT_VERIFYPASSWORD 0x13
#define FINGERPRINT_HISPEEDSEARCH 0x1B
[... 355 lines cut ...]
          (reply[1] != (FINGERPRINT_STARTCODE & 0xFF)))
          return FINGERPRINT_BADPACKET;
      uint8_t packettype = reply[6];
      //Serial.print("Packet type"); Serial.println(packettype);
      uint16_t len = reply[7];
      len <<= 8;
      len |= reply[8];
      len -= 2;
      //Serial.print("Packet len"); Serial.println(len);
      if (idx <= (len+10)) continue;
      packet[0] = packettype;      
      for (uint8_t i=0; i<len; i++) {
        packet[1+i] = reply[9+i];
      }
#ifdef FINGERPRINT_DEBUG
      Serial.println();
#endif
      return len;
    }
  }
}

This code will work even in the new version of the Arduino IDE and ESP8266. While working with esp serial communication pins has to be connected in 14 and 12 pins 

SoftwareSerial mySerial(14, 12,false,128);

suryapraneeta, 2017-10-12:
For this unknown error we need to just rewrite the header file and .cpp file. the data reading from the sensor has to be stored in a buffer.
--------------------------------------------------------------------------------
Header file has to be modified as 
/*************************************************** 
  This is a library for our optical Fingerprint sensor

  Designed specifically to work with the Adafruit Fingerprint sensor
  ----> http://www.adafruit.com/products/751

  These displays use TTL Serial to communicate, 2 pins are required to 
  interface
  Adafruit invests time and resources providing this open source code, 
  please support Adafruit and open-source hardware by purchasing 
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.  
  BSD license, all text above must be included in any redistribution
 ****************************************************/

#if (ARDUINO >= 100)
 #include "Arduino.h"
 #include <SoftwareSerial.h>
#else
 #include "WProgram.h"
 #include <NewSoftSerial.h>
#endif

#define FINGERPRINT_OK 0x00
#define FINGERPRINT_PACKETRECIEVEERR 0x01
#define FINGERPRINT_NOFINGER 0x02
#define FINGERPRINT_IMAGEFAIL 0x03
#define FINGERPRINT_IMAGEMESS 0x06
#define FINGERPRINT_FEATUREFAIL 0x07
#define FINGERPRINT_NOMATCH 0x08
#define FINGERPRINT_NOTFOUND 0x09
#define FINGERPRINT_ENROLLMISMATCH 0x0A
#define FINGERPRINT_BADLOCATION 0x0B
#define FINGERPRINT_DBRANGEFAIL 0x0C
#define FINGERPRINT_UPLOADFEATUREFAIL 0x0D
#define FINGERPRINT_PACKETRESPONSEFAIL 0x0E
#define FINGERPRINT_UPLOADFAIL 0x0F
#define FINGERPRINT_DELETEFAIL 0x10
#define FINGERPRINT_DBCLEARFAIL 0x11
#define FINGERPRINT_PASSFAIL 0x13
#define FINGERPRINT_INVALIDIMAGE 0x15
#define FINGERPRINT_FLASHERR 0x18
#define FINGERPRINT_INVALIDREG 0x1A
#define FINGERPRINT_ADDRCODE 0x20
#define FINGERPRINT_PASSVERIFY 0x21

#define FINGERPRINT_STARTCODE 0xEF01

#define FINGERPRINT_COMMANDPACKET 0x1
#define FINGERPRINT_DATAPACKET 0x2
#define FINGERPRINT_ACKPACKET 0x7
#define FINGERPRINT_ENDDATAPACKET 0x8

#define FINGERPRINT_TIMEOUT 0xFF
#define FINGERPRINT_BADPACKET 0xFE

#define FINGERPRINT_GETIMAGE 0x01
#define FINGERPRINT_IMAGE2TZ 0x02
#define FINGERPRINT_REGMODEL 0x05
#define FINGERPRINT_STORE 0x06
#define FINGERPRINT_LOAD 0x07
#define FINGERPRINT_UPLOAD 0x08
#define FINGERPRINT_DELETE 0x0C
#define FINGERPRINT_EMPTY 0x0D
#define FINGERPRINT_VERIFYPASSWORD 0x13
#define FINGERPRINT_HISPEEDSEARCH 0x1B
[... 354 lines cut ...]
          (reply[1] != (FINGERPRINT_STARTCODE & 0xFF)))
          return FINGERPRINT_BADPACKET;
      uint8_t packettype = reply[6];
      //Serial.print("Packet type"); Serial.println(packettype);
      uint16_t len = reply[7];
      len <<= 8;
      len |= reply[8];
      len -= 2;
      //Serial.print("Packet len"); Serial.println(len);
      if (idx <= (len+10)) continue;
      packet[0] = packettype;      
      for (uint8_t i=0; i<len; i++) {
        packet[1+i] = reply[9+i];
      }
#ifdef FINGERPRINT_DEBUG
      Serial.println();
#endif
      return len;
    }
  }
}

This code will work even in the new version of the Arduino IDE and ESP8266. While working with esp serial communication pins has to be connected in 14 and 12 pins 

SoftwareSerial mySerial(14, 12,false,128);

nikay-20, 2017-10-18:
Did not find fingerprint sensor. i am using arduino mega, how to solve this?

nikay-20, 2017-10-18:
how to fix the unknown error

ladyada, 2017-11-25:
hiya you may want to try the latest version we just committed. it fixes many bugs - thx!

closed 2017-11-25 by ladyada

slippy2016, 2018-02-18:
hi can some one i am trying to enrol and the it can not find the finger reader

dhalbert, 2018-02-18:
@slippy2016 Hi - For support, ask in https://forums.adafruit.com.

ZEYER1992, 2018-02-21:
IT IS ONLY THE VERSION OF ARDUINO IDE, TRY WHIT ARDUINO IDE 1.6.7 IT WORKS¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡¡

ARBAB1, 2019-12-16:
> I rewrote the library as a work around to the 1.8.0 problem, because it definitely WAS NOT FIXED. There were still problems as I mentioned above, based on the sketch.
>
> To correct the problem I added [...]

sulemanzain1, 2019-12-16:
> The latest compilers are stable, they just have errors when compiling this type of dynamically declared class array. Switching back to lower versions of the compiler is not an option for people that [...]

sulemanzain1, 2019-12-16:
@rengelking can I get .h File, I couldn't understand how you update .h file

siddhu1052, 2023-11-05:
Hello, I am using R307 sensor with adafruit fingerprint sensor library. I t is enrolling the fingerprints but when I am running fingerprint example of library it shows "unknown error". I is unable to match the fingerprints.