adafruit/Adafruit-Fingerprint-Sensor-Library #17: MKR1000 with Fingerprint Sensor

chinmoyrick, 2016-11-02:
I want to connect my fingerprint device with MKR1000. with hardware serial pin13-RX and 14-TX

As I connect this fingerprint device with arduino MICRO .With this sketch, It working with Hardware serial. In this sketch some instruction how to connect with micro hardware serial its working fine . 

Then i try to upload this sketch with my new MKR1000 module . i remove some line from main sketch those are.
`#include <SoftwareSerial.h>
// pin #2 is IN from sensor (GREEN wire)
// pin #3 is OUT from arduino  (WHITE wire)
SoftwareSerial mySerial(2, 3);
Adafruit_Fingerprint finger = Adafruit_Fingerprint(&mySerial);`
and enter those line for micro and also MKR1000
`//On Leonardo/Micro or others with hardware serial, use those! #0 is green wire, #1 is white
Adafruit_Fingerprint finger = Adafruit_Fingerprint(&Serial1);`

Doing those step its successfully upload to micro .. but its return some error with MKR100 

Adafruit_Fingerprint.h file and Adafruit_Fingerprint.cpp file have this softwareserial `#include <SoftwareSerial.h>`
and more related file for connecting software serial connection , for this there is uploading error. 

**_Build options changed, rebuilding all
In file included from C:\Users\sony\Documents\Arduino\Adafruit-Fingerprint-Sensor-Library-master\examples\fingerprint\fingerprint.ino:18:0:
C:\Users\sony\Documents\Arduino\libraries\Adafruit-Fingerprint-Sensor-Library/Adafruit_Fingerprint.h:18:28: fatal error: SoftwareSerial.h: No such file or directory
 #include <SoftwareSerial.h>
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino MKR1000._**
 those  error come at compile time ..
Thank u

chinmoyrick, 2016-11-02:
After changing , remove some line from .cpp and .h file 
related code with software serial and along with   #include <SoftwareSerial.h>
then i compile and run with my arduino micro ,its working without any error
and when upload the same code with micro error is
**C:\Users\sony\Documents\Arduino\Adafruit-Fingerprint-Sensor-Library-master\examples\fingerprint\fingerprint.ino:20:24: fatal error: util/delay.h: No such file or directory
 #include <util/delay.h>
                        ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino MKR1000.**

Same issue like https://github.com/adafruit/MAX6675-library/issues/9

ladyada, 2017-11-25:
hiya you may want to try the latest version we just committed. it fixes many bugs - use hardware serial rather than software. we just tested it with a feather m0 successfully :)

closed 2017-11-25 by ladyada