adafruit/Adafruit-Fingerprint-Sensor-Library #19: Is not working with Arduino IDE 1.6.13!  But with Arduino IDE 1.6.7 :)=

Grunewald17, 2016-12-17:
I was wondering why the fingersensor is not working. It was working the past years. 

After searching for hours... and testing... and thinking... and testing... I found the reason why.

The Arduino IDE 1.6.13 has one issue with arrays. 

If i use ...   byte[] arr

for(i=0;i<5;i++) Serial.print(arr[i]); 

all is working fine.  But...

byte test = arr[1];  .... then I will get wrong values.

GrexAut, 2017-01-01:
Look at https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/pull/18 it is working with latest Arduino IDE (1.8.0)

ladyada, 2017-11-25:
hiya you may want to try the latest version we just committed. it fixes many bugs - thx!

closed 2017-11-25 by ladyada