adafruit/Adafruit-Fingerprint-Sensor-Library #25: when this library is added to eclipse IDE,i am getting a error "Multiple makers at this line,redefinition of class Adafruit_Fingerprint,previous definition of cass Adafruit_fingerprint".path are builded correctly as done always .what is solution to this error"

dinesh2895, 2017-03-03:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

M-ax-G, 2017-07-10:
Have the same issue... further research reveals that include guards seem to be required to prevent the issue.

I added
`
#ifndef Adafruit_Fingerprint
#define Adafruit_Fingerprint
`
at the beginning of
Adafruit_Fingerprint.h
and `#endif` at the end.
Unfortunately it worsened the error message.

The code compiles with the Arduino IDE but not with the Arduino plug-in for the sloeber IDE.

A similar issue has been encountered in different modules of the Adafruit libraries; e.g. here:
https://github.com/adafruit/Adafruit_SSD1306/pull/27

Unfortunately I do not know enough C to fix the problem.

Here the sloeber compile result:
`0:40:22 **** Incremental Build of configuration Release for project FingerprintReader ****
"D:\\Prog\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../.ino.cpp'
'Starting C++ compile'
"D:\Prog\sloeber\arduinoPlugin\tools\arduino\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"D:\Prog\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.16\cores\arduino" -I"D:\Prog\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.16\variants\eightanaloginputs" -I"D:\Prog\sloeber\arduinoPlugin\libraries_added\Adafruit-Fingerprint" -I"D:\Prog\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.16\libraries\SoftwareSerial" -I"D:\Prog\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.16\libraries\SoftwareSerial\src" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp" -o ".ino.cpp.o"  -Wall
In file included from ../FingerprintReader.ino:21:0,
                 from ../.ino.cpp:15:
D:\Prog\sloeber\arduinoPlugin\libraries_added\Adafruit-Fingerprint/Adafruit_Fingerprint.h:72:7: error: redefinition of 'class Adafruit_Fingerprint'
 class Adafruit_Fingerprint {
       ^
In file included from ../.ino.cpp:8:0:
D:\Prog\sloeber\arduinoPlugin\libraries_added\Adafruit-Fingerprint/Adafruit_Fingerprint.h:72:7: error: previous definition of 'class Adafruit_Fingerprint'
 class Adafruit_Fingerprint {
       ^
make: *** [subdir.mk:26: .ino.cpp.o] Error 1

20:40:22 Build Finished (took 557ms)
`
[image]

ladyada, 2017-11-27:
we've added headerguards, should be resolved!

closed 2017-11-27 by ladyada