adafruit/Adafruit-Fingerprint-Sensor-Library #27: getTemplateCount always returns 0

SamTebbs33, 2017-08-20:
- Arduino board:  Uno WiFi DevEd

- Arduino IDE version (found in Arduino -> About Arduino menu): 1.8.3

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Run enroll example sketch, then run following code:

```
#include <Adafruit_Fingerprint.h>

void setup() {
  Adafruit_Fingerprint f = Adafruit_Fingerprint(&SoftwareSerial(2, 3));
  f.begin(57600);
  while(!Serial);
  Serial.begin(9600);
  Serial.println(f.getTemplateCount());
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

This always prints 0, even if there are templates stored.

tcosmo, 2017-11-08:
Having the same issue here!

tcosmo, 2017-11-08:
Ok you should solve it as follows:
`  finger.getTemplateCount();
  Serial.println(finger.templateCount);`

All the functions of the library returns error codes, 0 meaning everything went well.
The value we want is in fact stocked in finger.templateCount.

ladyada, 2017-11-28:
clarified in fingerprint.ino demo
thx!

closed 2017-11-28 by ladyada