adafruit/Adafruit-Fingerprint-Sensor-Library #28: Enroll Finger print creates issues; Adafruit Library

vindhyachaltakniki, 2017-09-07:
1. I am using Uno, finger print module & Adafruit library.
 
2. With using all the library all the function works ok, like delete, scan but i have some issues in Enroll function.
 
3. Enroll function is below, it is same, except some minor changes & I had a small key_read function, in case user wants to exit. key_read is small non-blocking function, read a variable only, which get set/reset in ISR only, so minimal code there.
 
4. Problem is :
a) When first scan is done, it is ok.
b) next it ask for remove finger & place again.
c) now doing that & place finger again, code many times gives finger mismatch.
d) 9 out of 10 times it is giving mismatch . Although I am doing the same way every-time.
 
5. Same happens if I use En-roll example from Adafruit also.
 
6. What could be the error?
 
7. One possible error I think in Adafruit library there is variable "confidence", which check the matching level of both. Might be it is set to very high value, but in library I didn't find where to set it low.

In Enroll function, it causes problem in createmodel function , it reurns mismatch.

`
uint8_t add_fingerprint_id(uint16_t id , uint8_t *r_key)
{
    uint8_t point = 255U;
    uint8_t key = no_key;
    uint8_t breakloop;
    uint8_t next = 0U;
  
    lcd.setCursor(0 , 3);   
    lcd_string_const(add_place_str); 
    serial_string_nl_const(add_place_str);     
    

/* clear key vars */
    clear_key_vars(); 

    breakloop = 0U;
    while(0U == breakloop)
    {
        point = finger.getImage();
        switch(point) 
        {
            case FINGERPRINT_OK:
                lcd.setCursor(0 , 3); 
                lcd_string_const(add_taken_str); 
                serial_string_nl_const(add_taken_str);
                breakloop = 1U;
            break;
    
            case FINGERPRINT_NOFINGER:
                lcd.setCursor(0 , 3); 
                lcd_string_const(add_no_finger_str);
                serial_string_nl_const(add_no_finger_str);
            break;
    
            case FINGERPRINT_PACKETRECIEVEERR:
                lcd.setCursor(0 , 3); 
                lcd_string_const(add_comm_err_str);
                serial_string_nl_const(add_comm_err_str);
            break;
    
            case FINGERPRINT_IMAGEFAIL:
                lcd.setCursor(0 , 3); 
                lcd_string_const(add_image_err_str);
                serial_string_nl_const(add_image_err_str);
            break;
    
            default:
                lcd.setCursor(0 , 3); 
                lcd_string_const(add_unknown_err_str);
[... 288 lines cut ...]
        key = key_read();
        if(EXIT_KEY == key)
        {
            breakloop = 1U; 
            next = 1U;          
        }                
    }      

    
    
    if(EXIT_KEY == key)
    {
        *r_key = EXIT_KEY;  
    }   
    else
    {
        *r_key = no_key;        
    }

    delay(2000);

    return point;
     
} /* function ends here */ 
`

ladyada, 2017-11-25:
hiya you may want to try the latest version we just committed. it fixes many bugs - thx!

closed 2017-11-25 by ladyada

sivabharaniii, 2019-02-08:
Hello vindhyachal at present i am also facing the same issue what you have mentioned,if you fixed the above problem please let me know.

sivabharaniii, 2019-02-11:
I am also facing the same issue as mentioned above,adafruit want to respond immediately on these  one.

caternuson, 2019-02-13:
I just tested this and can't recreate the issue. I wired the sensor to an UNO as shown here:
https://learn.adafruit.com/adafruit-optical-fingerprint-sensor/wiring-for-use-with-arduino
and ran the `enroll` example for the library (version 1.1.2). 

[image]

The only change I made was to comment out the `Serial.print('.');` lines since they were scrolling the serial console too much. I ran it several times and tried different fingers, but it seems to work each time:

[image]