adafruit/Adafruit-Fingerprint-Sensor-Library #36: Function to send templates from your controller to the sensor

ricarduscat, 2017-12-16:
`uint8_t Adafruit_Fingerprint::downloadModel(uint16_t id, uint8_t packet_1[], uint8_t packet_2[], uint8_t slot) {
  uint8_t packet[] = {FINGERPRINT_DOWNLOAD, slot};
  writePacket(theAddress, FINGERPRINT_COMMANDPACKET, sizeof(packet)+2, packet);
  uint8_t len = getReply(recvPacket);
  
  if ((len != 1) && (recvPacket[0] != FINGERPRINT_ACKPACKET))
   return -1;
  if (recvPacket[1] == 0x00) {
    writePacket(theAddress, FINGERPRINT_DATAPACKET, 130, packet_1);
    writePacket(theAddress, FINGERPRINT_DATAPACKET, 130, packet_2);
    return recvPacket[1];
  } else {
    return recvPacket[1];
  }
}`

I made this with the latest version of the code. I need to put templates back to the sensor when I free the memory. The documention of the sensor says it's allowed, but not implemented in the library. This does not work, it will be fine to improve the library.

marianodato, 2018-03-10:
This function worked for you? Did you find another solution?

ricarduscat, 2018-03-26:
Yes, but I had to touch the fingerprint sensor packet size to adapt it to the datapackets I was sending

marianodato, 2018-03-26:
@ricarduscat could you explain how to do that?

milindrc, 2018-09-27:
@ricarduscat please explain the parameter you need to pass to this function.

sand007man, 2018-09-28:
i am able to enroll ,search fingerprint but when i load show_fingperprint _template it is showing some "??" values in the serial monitor . even small help will be greatly appreciated . I am using this fingerprint with node_mcu esp 12e...i feel its something related to memory buffer !!

hanifizzudinrahman, 2019-02-21:
> recvPacket

what is "recvPacket"? the last packet, or what?

abhishekgautam101, 2019-04-09:
### please refer to my code
[https://github.com/abhishekgautam101/r30x-fingerprint-sensor-library-for-arduino](url)

 I am using the following code to push buffer 1 template(which I got by using upChar) from Arduino to fingerprint sensor
```void match(){
   while (!Serial.available());
   String str;
   for(int i=0; i< 147; i++){             //reading string from serial
    if(Serial.available()){
      str += Serial.read();
    }
   }
   
  byte byte_arr[63];
  int arr_element_index = 0;
  int startIndex = 0;
  int endIndex = 0;
  for(int i=0 ; i<str.length() ; i++){          //converting string to hex byte array
    if(str.charAt(i) == ','){
      endIndex = i;
      String str_part =  str.substring(startIndex, endIndex);
      char charBuf[2];
      str_part.toCharArray(charBuf, 2);
      byte_arr[arr_element_index] = (byte) strtoul(charBuf, NULL, 16);
      startIndex = endIndex + 1;
      arr_element_index++;
    }
  }

   r30x_fl.begin(115200);        //sending data to fingerprint module
   delay(1000);

   if(r30x_fl.verifyPassFunc()){
    if(r30x_fl.callGetImage()){
      if(r30x_fl.img2buffer2Func()){
        //feeding previous data of user to fingerprint to match
          if(r30x_fl.downCharBuffer1Func(byte_arr)){
            if(r30x_fl.matchBuffersFunc()){
              delay(1000);
               Serial.print("~Fingers mached!$");
            }else{
              delay(1000);
              Serial.print("~Fingers did not match! Try again...$");
            }
          }else{
            delay(1000);
            Serial.print("~Unknown error! Try again...$");
          }
      }else{
        delay(1000);
        Serial.print("~Finger not Captured properly! Try again...$");
      }
    }else{
      delay(1000);
      Serial.print("~Time Out! Try again...$");
    }
  }else{
    delay(1000);
    Serial.print("~Incorrect credential! Try again...$");
  }
}
```
the template buffer which I am sending is 
```EF,1,FF,FF,FF,FF,2,0,82,3,1,4D,1C,86,0,80,6,80,6,0,6,0,6,0,6,0,2,80,2,80,2,80,6,C0,6,C0,6,C0,6,E0,1E,F0,3E,FF,FE,FF,FE,FF,FE,0,0,0,0,0,0,0,0,18,18,BD,27,EE,8F,```
as a string , then decoding it in above function.

Please tell me what I am doing wrong!

jimmygyl, 2019-08-10:
> 
> 
> i am able to enroll ,search fingerprint but when i load show_fingperprint _template it is showing some "??" values in the serial monitor . even small help will be greatly appreciated . I am using this fingerprint with node_mcu esp 12e...i feel its something related to memory buffer !!

Try using larger buffer for the software serial
something like
SoftwareSerial mySerial(D7, D8, false, 1024);

shreeramk1998, 2020-01-17:
> ```
> uint8_t Adafruit_Fingerprint::downloadModel(uint16_t id, uint8_t packet_1[], uint8_t packet_2[], uint8_t slot) {
>   uint8_t packet[] = {FINGERPRINT_DOWNLOAD, slot};
>   writePacket(theAddress, [...]

If possible can you please expand on this code ?(What are the parameters..etc). Is there any other way where I can load the templates from one sensor to other sensor. BTW I am using R307.

hanifizzudinrahman, 2020-01-29:
> > ```
> > uint8_t Adafruit_Fingerprint::downloadModel(uint16_t id, uint8_t packet_1[], uint8_t packet_2[], uint8_t slot) {
> >   uint8_t packet[] = {FINGERPRINT_DOWNLOAD, slot};
> >   writePacket(theAddress, FINGERPRINT_COMMANDPACKET, sizeof(packet)+2, packet);
> >   uint8_t len = getReply(recvPacket);
> >   
> >   if ((len != 1) && (recvPacket[0] != FINGERPRINT_ACKPACKET))
> >    return -1;
> >   if (recvPacket[1] == 0x00) {
> >     writePacket(theAddress, FINGERPRINT_DATAPACKET, 130, packet_1);
> >     writePacket(theAddress, FINGERPRINT_DATAPACKET, 130, packet_2);
> >     return recvPacket[1];
> >   } else {
> >     return recvPacket[1];
> >   }
> > }
> > ```
> > 
> > 
> > I made this with the latest version of the code. I need to put templates back to the sensor when I free the memory. The documention of the sensor says it's allowed, but not implemented in the library. This does not work, it will be fine to improve the library.
> 
> @ricarduscat If possible can you please expand on this code ?(What are the parameters..etc). Is there any other way where I can load the templates from one sensor to other sensor. BTW I am using R307.

i suggest you to read this post
https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/issues/9

i have written on there, how to send template fingerprint to module fingerprint

adityadate1997, 2020-03-13:
I want to download and store the template in my server datatabase, I am using NodeMCU and Php with mysql on server side. This function is not provided in the library. Can anybody please guide me on this?

hanifizzudinrahman, 2020-03-14:
> I want to download and store the template in my server datatabase, I am using NodeMCU and Php with mysql on server side. This function is not provided in the library. Can anybody please guide me on this?

of course, i can help you. NodeMCU i think can use as Arduino, so it can be work with module fingerprint R307 or DY50

adityadate1997, 2020-03-14:
> of course, i can help you. NodeMCU i think can use as Arduino, so it can be work with module fingerprint R307 or DY50

Can we connect through an email? I'll send in the required code for you to see.

hanifizzudinrahman, 2020-03-14:
yes, you can see my profile to get my new phone number or email

adityadate1997, 2020-03-18:
Sir, we have downloaded the fingerprint template data from the module r305
using the show_finger_template library. It gives us an array of size
512....we want to compare this stored array on server side. What exactly do
we need to do? We tried comparing the array byte by byte but it does not
give proper output. Can you guide us on how to match these templates sir.

Thank you.

rodrigotabsan, 2020-06-11:
Hello adityadate1997, 

You need to store the template into your fingerprint sensor and compare it with the fingerprint that the sensor is receiving from the user. Probably, you will need to store templates from a database to the database of the sensor to check the fingerprint that your sensor has received and compare it with the fingerprints of the database.

Good luck

shreeramk1998, 2020-06-11:
@rodrigotabsan Are you able to get the templates out from the sensor and store those templates into another sensor's flash memory? If so what sensor are you using ?
I'm also trying to achieve the same thing using NodeMCU and R307.

rodrigotabsan, 2020-07-01:
@shreeramk1998 I didn't try that. I can't get the fingerprint template I have stored in an external database to match the one I put on the sensor. I'm not sure this can be done... I've been working on this for several months with no results.

6000850, 2021-04-15:
Hello
This is the 3-finger pattern output stored in the module
I have saved these templates in the server database. Can I restore these templates to another module (fingerprint) and be usable?

```
Attempting to load #1
Template 1 loaded
Attempting to get #1
Template 1 transferring:
534 bytes read.
Decoding packet...
FFFFFFFFFFFFFFFFFF09144BAE6B8A6845BEBE6E4D9D4272AA7B0C76BA77947A95EF9C7EB5E1A482ABC7D90DE993A30F54445703636883651396377E32A45E204D00EF0B091E0E7A7C98780A16A94E3047C3190CC0605F70DF82D390C61C921441F152FFBD1DDF57721022EA517970EE836D7086133DB05673935C1A141675D6504177F6000000003753EF01FFFFFFFF02008200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000FFFFFFFFFFFFFFFFFFFFFFD6EF01FFFFFFFF020082CB091455A6AB826425BE9E6E0D9D6272AA7B8C76BA77947A95EF9C7EB5E1A482ABC7590DEB973F378BC79A52D96E2F71B2907F7B42B8AD68EA6844BD85CE3C77C758AEEE5BA97F2ECAC12AD8CB7AFF74DDB03E1AA4F85C947949B472FE2C1F7F233F2D3C4A67EFCE93E63ECA302BB15A3516C04229C675D6C30F77F6000000004076EF01FFFFFFFF080082CB09144BAE6B8A6845BEBE6E4D9D4272AA7B0C76BA
done.
------------------------------------
Attempting to load #2
Template 2 loaded
Attempting to get #2
Template 2 transferring:
534 bytes read.
Decoding packet...
FFFFFFFFFFFFFFFFFF091852A1AA859405C6BE60CDAD0274AB5B0D74B937157556E72BC15598B7B4BA8439B2062970051716C200C3C9425E2D8D2AB7DFE33AD4A78F004DB69EA401FEF01FFFFFFFF020082DC95A39E80F4FAAE6BDCFEEE00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000FFFFFFFFFFFFFFFFFFFFFFAFEF01FFFFFFFF020082CB091B4FA26B826805B69E6E0D9D4272AA7B8C76BA77147A97EF1C7EB5E1A482ABC75909E9937B209045F75FF86F876003928D7C02A0445BA770BD45489EA73A5ED96F82C1797F80F8C38EEC44626FBC898E54BA20E0B59AAEC0722667BFB1E1B7DB3CC7D3EEFD5433683A26C94A6FCE05DA69D654287A3A06596BC785FCCBB74230EF01FFFFFFFF080082CB091852A1AA859405C6BE60CDAD0274AB5B0D74B9371578936F1D78B3E22584A7C95A11E7AB773FF847AE2C086D5A651390556E72BC1550EA68C44049DE07C1DB399AA64378E9D8CBC1DEC24E666D98B7B4BA8439FCF8
done.
Attempting to load #3
Template 3 loaded
Attempting to get #3
Template 3 transferring:
534 bytes read.
Decoding packet...
FFFFFFFFFFFFFFFFFF091A59BE4BFA6CC5A6FE6CCC8D8272A97B0D76B977157A93E5989C1802BC050F275855AF56A8FCBA692308EAB8C748CA430B384B17E3C87CF78BD7383E8629F68C2AE972222B5F230B8D5D634CCEFAEC69ADE5ABE1FFCFA41F0EF01FFFFFFFF0200825FFDFADD8565A01605E9A93E7318022FCED30C6F0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000FFFFFFFFFFFFFFFFFFFFFF8EEF01FFFFFFFF020082CB09175EAE2BBA6C45B6FE6CCD9D0272A97B0D76B977157A93EF1D7EB3E12582AFC55801EF9B3303F344E23D536F7E10239C541372B0B7746940F56E68EE7D3586992A6E9B2AA9504344DF464A68B8E2DF8E118CAA9432BECC28F79A804E3D83C738B80B3AE8FC1790407E70BB216FC28A564332FB3945BA191BDA4B669A54DE3D14EF01FFFFFFFF080082CB091A59BE4BFA6CC5A6FE6CCC8D8272A97B0D76B977157A93EF1D7CB3E02580A7C15A01E78B0F1F79C48608736FF63B95989C1802BC050F275854038
done.

```
please guide me
Thanks

hanifizzudinrahman, 2021-04-16:
Do it with example code "Upload",  so the template will move to another
module fingerprint

a2br, 2021-06-13:
> Can we connect through an email? I'll send in the required code for you to see.

I'd be really interested to find out how you did that. Could you please reach out to me or make what you learned public by any means?

mlinaje, 2021-09-29:
After reading all the datasheets I have found. Does someone knows if the template/signature of the fingerprint format follows some kind of standard such as ISO, ANSI,...?
Thanks in advance!

Dygear, 2022-06-20:
So, I read the datasheet, looks like this uses [the DownImage function documented here on page 19](https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/master/documentation/ZFM-20_Fingerprint_Module.pdf#page=19). You can also upload an image to the sensor with UpImage (found on page 18.)

The images themselves and the formatting used are documented on page 8 (under the title 4.1.1 Image Buffer) says that the captured images are 256x288 pixels in size.

> When uploading or downloading images though the UART port, in order to speed up the speed, only the upper four bits of the pixel byte are used that is, 16-level grayscale is used, and each byte represents two pixels (the high four bits are one pixel, and the low four bits are the same. A pixel in the next adjacent column of the row, that is to pixels are combined into one byte and transmitted). Since the image is16 grayscale levels, uploaded to PC When displaying (corresponding to BMP format), the grayscale should be expanded (expanded to 256 grayscale, i.e. 8bit bitmap format.)
> Transmission though the USB port is a full 8-bit pixel, that is, 256 gray levels.

What I think is needed is for the image to tz function Img2Tz page 19 again, after capturing / sending two images with that function it stores the first image in `CharBuffer1` and `CharBuffer2`. Then you can run the RegModel function (page 20) that takes the contents of `CharBuffer1` and `CharBuffer1` and tries to make a template model out of it.

On page 21 it looks like you can straight up take the template model (they are now calling it characteristics) and "upload" that to the host computer. On page 21 again it has the DownChar function that allows you to download into the characestirics buffers and then you can save them into the flash memory with the store function call.

I think the only functions you _really_ need are the UpChar, DownChar and Store functions in order to sync a database across multiple devices and have it return the same position ID. If they became out of sync you can always delete the database and copy one fingerprint reader to another using the Empty function (documented on page 24).

AsifKhan991, 2022-09-05:
I was also in need of such functionalities (to read, save and write back template data to similar sensor models) and thus came up with an update to the existing library. I have a pull request pending about this. Please checkout my [repository about this update](https://github.com/AsifKhan991/Adafruit-Fingerprint-Sensor-Library/tree/template_transfer_feature). Also checkout the [changes](https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/pull/116#issue-1360942013) I've made. 

However, keep in mind that writing template data from one model to another won't serve the purpose as different sensor models might use different template matching algorithms. 

Hope I was able to help!

Gitare420, 2022-09-06:
Hi, I'm having trouble pasting the template.

get.template gives me this

57,1,7E,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,F,0,2,0,83,0,0,30,C,F3,33,F3,CF,FF,FE,EF,FF,FB,FB,EE,EF,EA,AA,AA,A9,95,66,59,55,55,65,55,55,51,54,40,44,4,40,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,.
3E,8A,EC,DE,4E,8B,57,1E,33,9D,52,BE,58,21,3,7E,46,A3,42,9E,24,A7,50,1E,5B,AA,46,1E,2E,2C,4F,9E,33,B9,27,3E,33,C2,E,7E,3E,8F,0,FF,6B,14,19,5F,51,14,D7,9F,4D,A5,17,FF,21,B6,E5,3F,48,37,95,1F,67,BB,DD,BF,25,41,CE,7F,57,C2,C7,9F,40,C3,9,BF,4D,43,88,DF,3B,2B,52,FA,36,2C,69,78,26,C6,4F,1E,38,50,4D,3C,3E,C6,D,DA,2A,50,A4,18,4A,D1,7,78,3D,C9,4D,99,42,C9,C9,B9,5A,4B,8,B9,42,CE,C,39,.
46,51,9,99,59,CD,9F,F5,59,CE,8,55,2E,50,65,95,2E,D0,8E,D5,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,.
3,3,56,17,0,1,20,1,85,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,F,0,2,0,77,0,0,30,C,F3,33,F3,CF,FF,FE,EF,FF,EF,BB,BA,AA,AA,AA,AA,65,55,55,95,55,55,45,51,10,44,4,40,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,.

and I paste it in write_to_sensor like this

uint8_t fingerTemplate[512] = {3, 3, 58, 25, 14, 1, 57, 1, 7E, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, F, 0, 2, 0, 83, 0, 0, 30, C, F3, 33, F3, CF, FF, FE, EF, FF, FB, FB, EE, EF, EA, AA, AA, A9, 95, 66, 59, 55, 55, 65, 55, 55, 51, 54, 40, 44, 4, 40, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 3E, 8A, EC, DE, 4E, 8B, 57, 1E, 33, 9D, 52, BE, 58, 21, 3, 7E, 46, A3, 42, 9E, 24, A7, 50, 1E, 5B, AA, 46, 1E, 2E, 2C, 4F, 9E, 33, B9, 27, 3E, 33, C2, E, 7E, 3E, 8F, 0, FF, 6B, 14, 19, 5F, 51, 14, D7, 9F, 4D, A5, 17, FF, 21, B6, E5, 3F, 48, 37, 95, 1F, 67, BB, DD, BF, 25, 41, CE, 7F, 57, C2, C7, 9F, 40, C3, 9, BF, 4D, 43, 88, DF, 3B, 2B, 52, FA, 36, 2C, 69, 78, 26, C6, 4F, 1E, 38, 50, 4D, 3C, 3E, C6, D, DA, 2A, 50, A4, 18, 4A, D1, 7, 78, 3D, C9, 4D, 99, 42, C9, C9, B9, 5A, 4B, 8, B9, 42, CE, C, 39, .46, 51, 9, 99, 59, CD, 9F, F5, 59, CE, 8, 55, 2E, 50, 65, 95, 2E, D0, 8E, D5, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3, 3, 56, 17, 0, 1, 20, 1, 85, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, F, 0, 2, 0, 77, 0, 0, 30, C, F3, 33, F3, CF, FF, FE, EF, FF, EF, BB, BA, AA, AA, AA, AA, 65, 55, 55, 95, 55, 55, 45, 51, 10, 44, 4, 40, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};

and it throws me the error exponent has no digits

What can be? Thanks in advance :)

AsifKhan991, 2022-09-06:
> Hi, I'm having trouble pasting the template.
>
> get.template gives me this [...]

These are actually hex values. You need to put '0x' before the hex values. Like this: 0xFE

AsifKhan991, 2022-09-06:
@Gitare420 Did you use the same sensor for reading the template from and writing to back it? What is the model pf the sensor? Also what id you enrolled against?

Gitare420, 2022-09-06:
use the same model 071405 finger to read and write, with the id 1

Gitare420, 2022-09-06:
I am attaching my codes, I am using the example ones that the library brings, sorry for my bad English, I am using google translator, greetings from Chile :)

//////////////////////////////////// get_template ////////////////////////////////////
#include <Adafruit_Fingerprint.h>
Adafruit_Fingerprint finger = Adafruit_Fingerprint(&Serial2);

uint8_t f_buf[512]; //here is where the template data is sotred

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  finger.begin(57600);
  while (!Serial);  // For Yun/Leo/Micro/Zero/...
  delay(100);
  Serial.println("\n\nAdafruit finger detect test");
  
  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1) { delay(1); }
  }
  store_template_to_buf();
}

void store_template_to_buf(){

  Serial.println("Waiting for valid finger....");
  while (finger.getImage() != FINGERPRINT_OK) { // press down a finger take 1st image 
  }
  Serial.println("Image taken");

  if (finger.image2Tz(1) == FINGERPRINT_OK) { //creating the charecter file for 1st image 
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }

  Serial.println("Remove finger");
  delay(2000);
  uint8_t p = 0;
  while (p != FINGERPRINT_NOFINGER) {
    p = finger.getImage();
  }

  Serial.println("Place same finger again, waiting....");
  while (finger.getImage() != FINGERPRINT_OK) { // press the same finger again to take 2nd image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(2) == FINGERPRINT_OK) { //creating the charecter file for 2nd image 
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }

  Serial.println("Creating model...");

  if (finger.createModel() == FINGERPRINT_OK) {  //creating the template from the 2 charecter files and saving it to char buffer 1
    Serial.println("Prints matched!");
    Serial.println("Template created");
  } else {
    Serial.println("Template not build");
    return;
  }

  Serial.println("Attempting to get template..."); 
  if (finger.getModel() == FINGERPRINT_OK) {  //requesting sensor to transfer the template data to upper computer (this microcontroller)
[... 185 lines cut ...]

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);

  return finger.fingerID;
}

// returns -1 if failed, otherwise returns ID #
int getFingerprintIDez() {
  uint8_t p = finger.getImage();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.image2Tz();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.fingerFastSearch();
  if (p != FINGERPRINT_OK)  return -1;

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);
  return finger.fingerID;
}
//////////////////////////////////// fingerprint ////////////////////////////////////

AsifKhan991, 2022-09-06:
@Gitare420 in the "write_to_sensor" example you have to comment out or delete the line "memset(fingerTemplate, 0xff, 512);" if you've manually put the template data in the array buffer before this line. (as you have done). This line was simply done to initiate the array buffer when it was previously empty. So comment out his line and the process should work. :)

Gitare420, 2022-09-06:
I made the change and left everything as it is and it still doesn't work, it reads the template and uploads it without problems, but when reading the fingerprint it can't find it in the database.

It occurs to me perhaps to filter the data that is not necessary and save the template in a string, and when reading it see the percentage of coincidence, and if it is greater than 80%, for example, I would take it as good. I'm going to sleep for a while and when I wake up I'll do that test and let you know how it goes, thanks for answering :)

hanifizzudinrahman, 2022-09-06:
> I was also in need of such functionalities (to read, save and write back template data to similar sensor models) and thus came up with an update to the existing library. @Dygear what you said is correct. I have a pull request pending about this. Please checkout my [repository about this update](https://github.com/AsifKhan991/Adafruit-Fingerprint-Sensor-Library/tree/template_transfer_feature). Also checkout the [changes](https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/pull/116#issue-1360942013) I've made.
> 
> However, keep in mind that writing template data from one model to another won't serve the purpose as different sensor models might use different template matching algorithms.
> 
> Hope I was able to help!

yap, that's right

AsifKhan991, 2022-09-06:
> I made the change and left everything as it is and it still doesn't work, it reads the template and uploads it without problems, but when reading the fingerprint it can't find it in the database. [...]

Don't see any other mistake in your code. The 512 template data is already filtered, there's no data headers or peripherals. So no further filtering required. I tested this library with R307, ZFM60 v1.4, R305 (synochip AS606 chip) and the official adafruit ones. Perhaps your model 071405 doesn't support the write instruction or have different instruction code. 

Also fingerprint template matching doesn't work the way you said.

AsifKhan991, 2022-09-06:
@Gitare420  @hanifizzudinrahman I made some changes in the `write_template_to_sensor()` function. Check if it works now. Also currently I don't have access to the modules. Tomorrow I will have them and check again thoroughly.

Gitare420, 2022-09-06:
Hello, I updated the library and it's still the same, from what I've seen my model 071405 is the least used, it may be the reason why it doesn't work, these are the data that the fingerprint example gives me:

Reading sensor parameters
Status: 0x0
SysID: 0x0
Capacity: 300
Security level: 3
Device address: FFFFFFFF
Packet length: 128
Baud rate: 57600
Waiting for valid finger...
Sensor contains 1 templates

I think I'll buy the 307 model through AliExpress (it doesn't exist in Chile) and I'll keep trying there so as not to waste my time, I'll be attentive to the updates, thank you very much :)

Gitare420, 2022-09-06:
I leave the last test I did in case it helps someone :)

//////////////////////////////////// write_to_sensor ////////////////////////////////////

#include <Adafruit_Fingerprint.h>
Adafruit_Fingerprint finger = Adafruit_Fingerprint(&Serial2);

void setup() {
  Serial.begin(115200);
  finger.begin(57600); //set your sensor's baudrate
  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1);
  }
  write_template_data_to_sensor();
}

uint8_t readnumber(void) {
  uint8_t num = 0;

  while (num == 0) {
    while (! Serial.available());
    num = Serial.parseInt();
  }
  return num;
}

void write_template_data_to_sensor() {
  uint8_t fingerTemplate[512] = {0x3, 0x3, 0x50, 0x16, 0x0, 0x1, 0x20, 0x1, 0x7F, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x12, 0x0, 0x1, 0x0, 0x71, 0x0, 0x33, 0x33, 0x33, 0xC3, 0xFF, 0xF, 0xFF, 0xFB, 0xEF, 0xFB, 0xFF, 0xFE, 0xAA, 0xAA, 0xAA, 0xAA, 0x55, 0x55, 0x59, 0x95, 0x55, 0x45, 0x55, 0x11, 0x51, 0x44, 0x10, 0x4, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x4C, 0x94, 0xC4, 0x5E, 0x50, 0x1E, 0x6, 0x7E, 0x28, 0xAE, 0x26, 0xFE, 0x29, 0xB7, 0x4D, 0x7E, 0x29, 0xBF, 0x23, 0x1E, 0x34, 0xC2, 0xAC, 0x3E, 0x46, 0x9, 0x57, 0xBF, 0x5B, 0xAF, 0x9E, 0x3F, 0x4C, 0xB6, 0x47, 0xDF, 0x3B, 0x97, 0x1, 0xDC, 0x41, 0x97, 0xD8, 0x5C, 0x33, 0xB9, 0xCC, 0x1C, 0x45, 0x40, 0x5E, 0xDC, 0x37, 0x3D, 0x8A, 0x5D, 0x30, 0x1F, 0x92, 0x5A, 0x2B, 0xA1, 0x28, 0xDA, 0x51, 0x42, 0xB, 0xDA, 0x37, 0x36, 0x9, 0xFB, 0x4C, 0x42, 0xE, 0xB8, 0x3D, 0x2B, 0x14, 0xD9, 0x3B, 0x33, 0x49, 0x32, 0x3D, 0x2E, 0x51, 0xB1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x3, 0x3, 0x50, 0x17, 0x0, 0x1, 0x20, 0x1, 0x81, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x10, 0x0, 0x3, 0x0, 0x6D, 0x0, 0x33, 0xCC, 0x3F, 0x3, 0xFC, 0x3F, 0xFF, 0xEF, 0xFF, 0xFE, 0xFF, 0xAA, 0xAA, 0xAA, 0xAA, 0xA9, 0x95, 0x55, 0x55, 0x54, 0x55, 0x51, 0x51, 0x54, 0x40, 0x4, 0x44, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0};

  Serial.println("Ready to write template to sensor...");
  Serial.println("Enter the id to enroll against, i.e id (1 to 127)");
  int id = readnumber();
  if (id == 0) {// ID #0 not allowed, try again!
    return;
  }
  Serial.print("Writing template against ID #"); Serial.println(id);

  if (finger.write_template_to_sensor(fingerTemplate)) { //telling the sensor to download the template data to it's char buffer from upper computer (this microcontroller's "fingerTemplate" buffer)
    Serial.println("now writing to sensor...");
  } else {
    Serial.println("writing to sensor failed");
    return;
  }

  Serial.print("ID "); Serial.println(id);
  if (finger.storeModel(id) == FINGERPRINT_OK) { //saving the template against the ID you entered or manually set
    Serial.print("Successfully stored against ID#"); Serial.println(id);
  } else {
    Serial.println("Storing error");
    return ;
  }
}

void loop() {}

//////////////////////////////////// write_to_sensor ////////////////////////////////////

AsifKhan991, 2022-09-07:
@Gitare420 maybe so, cause today I checked and everything working fine on my end. Another thing, the 512 byte template data is already filtered, no further filtering required. And the fingerprint template matching doesn't work the way you mentioned earlier

AsifKhan991, 2022-09-08:
There were indeed some issues with template buffer size. Those are now fixed. @Gitare420 sorry to bother, but will you please let me know if the issue is now solved or not? (use the new "get_direct_template.ino" example to get the template)

Gitare420, 2022-09-09:
WORKS!!!!! Thank you very much, I have searched everywhere and your library was the best that worked for me, very simple and intuitive everything, really thank you very much, you are a genius, I enclose my code in case it helps someone

////////////////////////////////////////write_to_sensor////////////////////////////////////////

#include <Adafruit_Fingerprint.h>
Adafruit_Fingerprint finger = Adafruit_Fingerprint(&Serial2);

void setup() {
  Serial.begin(115200);
  finger.begin(57600); //set your sensor's baudrate
  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1);
  }
  write_template_data_to_sensor();
}

uint8_t readnumber(void) {
  uint8_t num = 0;

  while (num == 0) {
    while (! Serial.available());
    num = Serial.parseInt();
  }
  return num;
}

void write_template_data_to_sensor() {
  int template_buf_size = 512; //usually hobby grade sensors have 512 byte template data, watch datasheet to know the info

  uint8_t fingerTemplate[512] = {0x3, 0x3, 0x53, 0x1A, 0x0, 0x1, 0x20, 0x1, 0x79, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x10, 0x0, 0x7, 0x0, 0x73, 0x0, 0x0, 0xC3, 0x3, 0x3C, 0xCF, 0xFF, 0xCF, 0xCC, 0xFF, 0xFB, 0xFE, 0xEA, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 0xA9, 0x95, 0x55, 0x55, 0x45, 0x11, 0x15, 0x11, 0x11, 0x11, 0x4, 0x40, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
                                 0x4F, 0x92, 0x1, 0xFE, 0x5C, 0x98, 0x59, 0x5E, 0x60, 0xA4, 0x86, 0x3E, 0x31, 0xAC, 0x10, 0x9E, 0x22, 0xB4, 0xE6, 0x9E, 0x33, 0x38, 0x29, 0x5E, 0x44, 0xB8, 0x94, 0x3E, 0x30, 0xC1, 0xCF, 0xBE, 0x60, 0xAE, 0x8, 0x3C, 0x4E, 0xA5, 0x43, 0x7A, 0x52, 0xA6, 0x59, 0xFA, 0x3E, 0xAB, 0xD4, 0x7A, 0x39, 0xAC, 0xAA, 0xF8, 0x3A, 0xC2, 0xD, 0x98, 0x3E, 0x1F, 0x95, 0xB9, 0x61, 0xC2, 0xA2, 0x59, 0x3E, 0xC2, 0xCC, 0x19, 0x39, 0x1F, 0x1, 0x76, 0x35, 0x9F, 0x93, 0x16, 0x65, 0xBF, 0x9F, 0xD6, 0x11, 0x43, 0xA4, 0x16, 0x3F, 0xBC, 0x8D, 0xF7, 0x41, 0x3E, 0xB, 0x97, 0x64, 0x40, 0x8B, 0xD7, 0x29, 0xA7, 0x50, 0xD4, 0x1A, 0xC2, 0x50, 0x94, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
                                 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
                                 0x3, 0x3, 0x54, 0x17, 0x0, 0x1, 0x20, 0x1, 0x79, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x10, 0x0, 0xD, 0x0, 0x5D, 0x0, 0xC, 0x30, 0xCC, 0xF3, 0xF3, 0x3F, 0xFF, 0xFF, 0xEA, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 0x65, 0x54, 0x51, 0x51, 0x45, 0x44, 0x51, 0x4, 0x4, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
                                };

  Serial.println("Ready to write template to sensor...");
  Serial.println("Enter the id to enroll against, i.e id (1 to 127)");
  int id = readnumber();
  if (id == 0) {// ID #0 not allowed, try again!
    return;
  }
  Serial.print("Writing template against ID #"); Serial.println(id);

  if (finger.write_template_to_sensor(template_buf_size, fingerTemplate)) { //telling the sensor to download the template data to it's char buffer from upper computer (this microcontroller's "fingerTemplate" buffer)
    Serial.println("now writing to sensor...");
  } else {
    Serial.println("writing to sensor failed");
    return;
  }

  Serial.print("ID "); Serial.println(id);
  if (finger.storeModel(id) == FINGERPRINT_OK) { //saving the template against the ID you entered or manually set
    Serial.print("Successfully stored against ID#"); Serial.println(id);
  } else {
    Serial.println("Storing error");
    return ;
  }
}

void loop() {}

////////////////////////////////////////write_to_sensor////////////////////////////////////////

Gitare420, 2022-09-10:
Create an example to add a fingerprint and get your template. I use an esp32.

////////////////////////////////////////enroll+get_direct_template////////////////////////////////////////

#include <Adafruit_Fingerprint.h>

Adafruit_Fingerprint finger = Adafruit_Fingerprint(&Serial2);

uint8_t f_buf[512]; //here is where the template data is sotred
uint8_t id = 6;
char Button;
int Condition_Read_Finger = '1';

void store_template_to_buf() {

  Serial.println("Waiting for valid finger....");
  while (finger.getImage() != FINGERPRINT_OK) { // press down a finger take 1st image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(1) == FINGERPRINT_OK) { //creating the charecter file for 1st image
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }
  Serial.println("Remove finger");
  delay(2000);
  uint8_t p = 0;
  while (p != FINGERPRINT_NOFINGER) {
    p = finger.getImage();
  }

  Serial.println("Place same finger again, waiting....");
  while (finger.getImage() != FINGERPRINT_OK) { // press the same finger again to take 2nd image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(2) == FINGERPRINT_OK) { //creating the charecter file for 2nd image
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }

  Serial.println("Creating model...");
  if (finger.createModel() == FINGERPRINT_OK) {  //creating the template from the 2 charecter files and saving it to char buffer 1
    Serial.println("Prints matched!");
    Serial.println("Template created");
  } else {
    Serial.println("Template not build");
    return;
  }

  Serial.print("ID "); Serial.println(id);
  if (finger.storeModel(id) == FINGERPRINT_OK) {
    p = finger.storeModel(id); if (p == FINGERPRINT_OK) {
      Serial.println("Stored!");
    } else {
      Serial.println("Unknown error");
      return;
    }

    Serial.println("Attempting to get template...");
    if (finger.getModel() == FINGERPRINT_OK) {  //requesting sensor to transfer the template data to upper computer (this microcontroller)
      Serial.println("Transferring Template....");
    } else {
      Serial.println("Failed to transfer template");
      return;
    }

    if (finger.get_template_buffer(512, f_buf) == FINGERPRINT_OK) { //read the template data from sensor and save it to buffer f_buf
      Serial.println("Template data (comma sperated HEX):");
      for (int k = 0; k < (512 / finger.packet_len); k++) { //printing out the template data in seperate rows, where row-length = packet_length
        for (int l = 0; l < finger.packet_len; l++) {
          Serial.print("0x");
          Serial.print(f_buf[(k * finger.packet_len) + l], HEX);
          Serial.print(",");
        }
      }
    }
    delay(2000);
    Condition_Read_Finger = '1';
  }
}

void setup() {
  Serial.begin(115200);
  finger.begin(57600); //set your sensor's baudrate
  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1);
  }

  Serial.print("write \"t\" to the serial monitor followed by the \"id\"  ");
  Serial.println("EXAMPLE \"t21\" for id 21");

  Serial.println("write \"Y\" in the serial monitor to delete all fingerprints");
}

void loop()
{
  if (Condition_Read_Finger == '1') {
    getFingerprintID();
    delay(50); //don't ned to run this at full speed.
  }

  if (Serial.available() != 0) {
    Button = Serial.read();

    if (Button == 't') {
      Condition_Read_Finger = '0';
      id = Serial.parseInt();
      Serial.print("Enrolling ID #");
      Serial.println(id);
      store_template_to_buf();
    }

    if (Button == 'Y') {
      finger.emptyDatabase();
      Serial.println("Now database is empty :)");
    }
  }
}

uint8_t getFingerprintID() {
  uint8_t p = finger.getImage();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image taken");
      break;
    case FINGERPRINT_NOFINGER:
      return p;
  }

  // OK success!

  p = finger.image2Tz();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image converted");
      break;
  }

  // OK converted!
  p = finger.fingerSearch();
  if (p == FINGERPRINT_OK) {
    Serial.println("Found a print match!");
  } else if (p == FINGERPRINT_NOTFOUND) {
    Serial.println("Did not find a match");
    return p;
  } else {
    Serial.println("Unknown error");
    return p;
  }

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);

  return finger.fingerID;
}

// returns -1 if failed, otherwise returns ID #
int getFingerprintIDez() {
  uint8_t p = finger.getImage();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.image2Tz();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.fingerFastSearch();
  if (p != FINGERPRINT_OK)  return -1;

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);
  return finger.fingerID;
}
////////////////////////////////////////enroll+get_direct_template////////////////////////////////////////

Gitare420, 2022-09-10:
Create an example to add a fingerprint and get your template.I use an esp32

////////////////////////////////////////enroll+get_direct_template////////////////////////////////////////
#include <Adafruit_Fingerprint.h>

Adafruit_Fingerprint finger = Adafruit_Fingerprint(&Serial2);

uint8_t f_buf[512]; //here is where the template data is sotred
uint8_t id = 6;
char Button;
int Condition_Read_Finger = '1';

void store_template_to_buf() {

  Serial.println("Waiting for valid finger....");
  while (finger.getImage() != FINGERPRINT_OK) { // press down a finger take 1st image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(1) == FINGERPRINT_OK) { //creating the charecter file for 1st image
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }
  Serial.println("Remove finger");
  delay(2000);
  uint8_t p = 0;
  while (p != FINGERPRINT_NOFINGER) {
    p = finger.getImage();
  }

  Serial.println("Place same finger again, waiting....");
  while (finger.getImage() != FINGERPRINT_OK) { // press the same finger again to take 2nd image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(2) == FINGERPRINT_OK) { //creating the charecter file for 2nd image
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }

  Serial.println("Creating model...");
  if (finger.createModel() == FINGERPRINT_OK) {  //creating the template from the 2 charecter files and saving it to char buffer 1
    Serial.println("Prints matched!");
    Serial.println("Template created");
  } else {
    Serial.println("Template not build");
    return;
  }

  Serial.print("ID "); Serial.println(id);
  if (finger.storeModel(id) == FINGERPRINT_OK) {
    p = finger.storeModel(id); if (p == FINGERPRINT_OK) {
      Serial.println("Stored!");
    } else {
      Serial.println("Unknown error");
      return;
    }

    Serial.println("Attempting to get template...");
    if (finger.getModel() == FINGERPRINT_OK) {  //requesting sensor to transfer the template data to upper computer (this microcontroller)
      Serial.println("Transferring Template....");
    } else {
      Serial.println("Failed to transfer template");
      return;
    }

    if (finger.get_template_buffer(512, f_buf) == FINGERPRINT_OK) { //read the template data from sensor and save it to buffer f_buf
      Serial.println("Template data (comma sperated HEX):");
      for (int k = 0; k < (512 / finger.packet_len); k++) { //printing out the template data in seperate rows, where row-length = packet_length
        for (int l = 0; l < finger.packet_len; l++) {
          Serial.print("0x");
          Serial.print(f_buf[(k * finger.packet_len) + l], HEX);
          Serial.print(",");
        }
      }
    }
    delay(2000);
    Condition_Read_Finger = '1';
  }
}

void setup() {
  Serial.begin(115200);
  finger.begin(57600); //set your sensor's baudrate
  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1);
  }

  Serial.print("write \"t\" to the serial monitor followed by the \"id\"  ");
  Serial.println("EXAMPLE \"t21\" for id 21");

  Serial.println("write \"Y\" in the serial monitor to delete all fingerprints");
}

void loop()
{
  if (Condition_Read_Finger == '1') {
    getFingerprintID();
    delay(50); //don't ned to run this at full speed.
  }

  if (Serial.available() != 0) {
    Button = Serial.read();

    if (Button == 't') {
      Condition_Read_Finger = '0';
      id = Serial.parseInt();
      Serial.print("Enrolling ID #");
      Serial.println(id);
      store_template_to_buf();
    }

    if (Button == 'Y') {
      finger.emptyDatabase();
      Serial.println("Now database is empty :)");
    }
  }
}

uint8_t getFingerprintID() {
  uint8_t p = finger.getImage();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image taken");
      break;
    case FINGERPRINT_NOFINGER:
      return p;
  }

  // OK success!

  p = finger.image2Tz();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image converted");
      break;
  }

  // OK converted!
  p = finger.fingerSearch();
  if (p == FINGERPRINT_OK) {
    Serial.println("Found a print match!");
  } else if (p == FINGERPRINT_NOTFOUND) {
    Serial.println("Did not find a match");
    return p;
  } else {
    Serial.println("Unknown error");
    return p;
  }

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);

  return finger.fingerID;
}

// returns -1 if failed, otherwise returns ID #
int getFingerprintIDez() {
  uint8_t p = finger.getImage();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.image2Tz();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.fingerFastSearch();
  if (p != FINGERPRINT_OK)  return -1;

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);
  return finger.fingerID;
}
////////////////////////////////////////enroll+get_direct_template////////////////////////////////////////

Gitare420, 2022-09-10:
Create an example to add a fingerprint and get your template. I use an esp32

////////////////////////////////////////enroll+get_direct_template////////////////////////////////////////
`#include <Adafruit_Fingerprint.h>

Adafruit_Fingerprint finger = Adafruit_Fingerprint(&Serial2);

uint8_t f_buf[512]; //here is where the template data is sotred
uint8_t id = 6;
char Button;
int Condition_Read_Finger = '1';

void store_template_to_buf() {

  Serial.println("Waiting for valid finger....");
  while (finger.getImage() != FINGERPRINT_OK) { // press down a finger take 1st image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(1) == FINGERPRINT_OK) { //creating the charecter file for 1st image
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }
  Serial.println("Remove finger");
  delay(2000);
  uint8_t p = 0;
  while (p != FINGERPRINT_NOFINGER) {
    p = finger.getImage();
  }

  Serial.println("Place same finger again, waiting....");
  while (finger.getImage() != FINGERPRINT_OK) { // press the same finger again to take 2nd image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(2) == FINGERPRINT_OK) { //creating the charecter file for 2nd image
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }

  Serial.println("Creating model...");
  if (finger.createModel() == FINGERPRINT_OK) {  //creating the template from the 2 charecter files and saving it to char buffer 1
    Serial.println("Prints matched!");
    Serial.println("Template created");
  } else {
    Serial.println("Template not build");
    return;
  }

  Serial.print("ID "); Serial.println(id);
  if (finger.storeModel(id) == FINGERPRINT_OK) {
    p = finger.storeModel(id); if (p == FINGERPRINT_OK) {
      Serial.println("Stored!");
    } else {
      Serial.println("Unknown error");
      return;
    }

    Serial.println("Attempting to get template...");
    if (finger.getModel() == FINGERPRINT_OK) {  //requesting sensor to transfer the template data to upper computer (this microcontroller)
      Serial.println("Transferring Template....");
    } else {
      Serial.println("Failed to transfer template");
      return;
    }

    if (finger.get_template_buffer(512, f_buf) == FINGERPRINT_OK) { //read the template data from sensor and save it to buffer f_buf
      Serial.println("Template data (comma sperated HEX):");
      for (int k = 0; k < (512 / finger.packet_len); k++) { //printing out the template data in seperate rows, where row-length = packet_length
        for (int l = 0; l < finger.packet_len; l++) {
          Serial.print("0x");
          Serial.print(f_buf[(k * finger.packet_len) + l], HEX);
          Serial.print(",");
        }
      }
    }
    delay(2000);
    Condition_Read_Finger = '1';
  }
}

void setup() {
  Serial.begin(115200);
  finger.begin(57600); //set your sensor's baudrate
  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1);
  }

  Serial.print("write \"t\" to the serial monitor followed by the \"id\"  ");
  Serial.println("EXAMPLE \"t21\" for id 21");

  Serial.println("write \"Y\" in the serial monitor to delete all fingerprints");
}

void loop()
{
  if (Condition_Read_Finger == '1') {
    getFingerprintID();
    delay(50); //don't ned to run this at full speed.
  }

  if (Serial.available() != 0) {
    Button = Serial.read();

    if (Button == 't') {
      Condition_Read_Finger = '0';
      id = Serial.parseInt();
      Serial.print("Enrolling ID #");
      Serial.println(id);
      store_template_to_buf();
    }

    if (Button == 'Y') {
      finger.emptyDatabase();
      Serial.println("Now database is empty :)");
    }
  }
}

uint8_t getFingerprintID() {
  uint8_t p = finger.getImage();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image taken");
      break;
    case FINGERPRINT_NOFINGER:
      return p;
  }

  // OK success!

  p = finger.image2Tz();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image converted");
      break;
  }

  // OK converted!
  p = finger.fingerSearch();
  if (p == FINGERPRINT_OK) {
    Serial.println("Found a print match!");
  } else if (p == FINGERPRINT_NOTFOUND) {
    Serial.println("Did not find a match");
    return p;
  } else {
    Serial.println("Unknown error");
    return p;
  }

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);

  return finger.fingerID;
}

// returns -1 if failed, otherwise returns ID #
int getFingerprintIDez() {
  uint8_t p = finger.getImage();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.image2Tz();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.fingerFastSearch();
  if (p != FINGERPRINT_OK)  return -1;

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);
  return finger.fingerID;
}`
////////////////////////////////////////enroll+get_direct_template////////////////////////////////////////

Gitare420, 2022-09-10:
Create an example to add fingerprints and get the templates at the same time. I use an esp32.

////////////////////////////////////////enrol+get_direct_template////////////////////////////////////////
#include <Adafruit_Fingerprint.h>

Adafruit_Fingerprint finger = Adafruit_Fingerprint(&Serial2);

uint8_t f_buf[512]; //here is where the template data is sotred
uint8_t id = 6;
char Button;
int Condition_Read_Finger = '1';

void store_template_to_buf() {

  Serial.println("Waiting for valid finger....");
  while (finger.getImage() != FINGERPRINT_OK) { // press down a finger take 1st image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(1) == FINGERPRINT_OK) { //creating the charecter file for 1st image
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }
  Serial.println("Remove finger");
  delay(2000);
  uint8_t p = 0;
  while (p != FINGERPRINT_NOFINGER) {
    p = finger.getImage();
  }

  Serial.println("Place same finger again, waiting....");
  while (finger.getImage() != FINGERPRINT_OK) { // press the same finger again to take 2nd image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(2) == FINGERPRINT_OK) { //creating the charecter file for 2nd image
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }

  Serial.println("Creating model...");
  if (finger.createModel() == FINGERPRINT_OK) { //creating the template from the 2 charecter files and saving it to char buffer 1
    Serial.println("Prints matched!");
    Serial.println("Template created");
  } else {
    Serial.println("Template not build");
    return;
  }

  Serial.print("ID "); Serial.println(id);
  if (finger.storeModel(id) == FINGERPRINT_OK) {
    p = finger.storeModel(id); if (p == FINGERPRINT_OK) {
      Serial.println("Stored!");
    } else {
      Serial.println("Unknown error");
      return;
    }

    Serial.println("Attempting to get template...");
    if (finger.getModel() == FINGERPRINT_OK) {  //requesting sensor to transfer the template data to upper computer (this microcontroller)
      Serial.println("Transferring Template....");
    } else {
      Serial.println("Failed to transfer template");
      return;
    }

    if (finger.get_template_buffer(512, f_buf) == FINGERPRINT_OK) { //read the template data from sensor and save it to buffer f_buf
      Serial.println("Template data (comma sperated HEX):");
      for (int k = 0; k < (512 / finger.packet_len); k++) { //printing out the template data in seperate rows, where row-length = packet_length
        for (int l = 0; l < finger.packet_len; l++) {
          Serial.print("0x");
          Serial.print(f_buf[(k * finger.packet_len) + l], HEX);
          Serial.print(",");
        }
      }
    }
    delay(2000);
    Condition_Read_Finger = '1';
  }
}

void setup() {
  Serial.begin(115200);
  finger.begin(57600); //set your sensor's baudrate
  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1);
  }

  Serial.print("write 't' to the serial monitor followed by the 'id' ");
  Serial.println("EXAMPLE 't21' for id 21");

  Serial.println("write 'Y' in the serial monitor to delete all fingerprints");
}

void loop()
{
  if (Condition_Read_Finger == '1') {
    getFingerprintID();
    delay(50); //don't ned to run this at full speed.
  }

  if (Serial.available() != 0) {
    Button = Serial.read();

    if (Button == 't') {
      Condition_Read_Finger = '0';
      id = Serial.parseInt();
      Serial.print("Enrolling ID #");
      Serial.println(id);
      store_template_to_buf();
    }

    if (Button == 'Y') {
      finger.emptyDatabase();
      Serial.println("Now database is empty :)");
    }
  }
}

uint8_t getFingerprintID() {
  uint8_t p = finger.getImage();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image taken");
      break;
    case FINGERPRINT_NOFINGER:
      return p;
  }

  // OK success!

  p = finger.image2Tz();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image converted");
      break;
  }

  // OK converted!
  p = finger.fingerSearch();
  if (p == FINGERPRINT_OK) {
    Serial.println("Found a print match!");
  } else if (p == FINGERPRINT_NOTFOUND) {
    Serial.println("Did not find a match");
    return p;
  } else {
    Serial.println("Unknown error");
    return p;
  }

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);

  return finger.fingerID;
}

// returns -1 if failed, otherwise returns ID #
int getFingerprintIDez() {
  uint8_t p = finger.getImage();
  if (p != FINGERPRINT_OK) return -1;

  p = finger.image2Tz();
  if (p != FINGERPRINT_OK) return -1;

  p = finger.fingerFastSearch();
  if (p != FINGERPRINT_OK) return -1;

  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID);
  Serial.print(" with confidence of "); Serial.println(finger.confidence);
  return finger.fingerID;
}
////////////////////////////////////////enrol+get_direct_template////////////////////////////////////////

AsifKhan991, 2022-09-10:
> Create an example to add fingerprints and get the templates at the same time. I use an esp32. [...]

it's better to keep your work in your repository and share the repository link in the comment section.

Gitare420, 2022-09-10:
Sorry, I made an account here 4 days ago and I'm learning how this works :)

Gitare420, 2022-09-12:
Hello, sorry for so many doubts, this is the last one hahaha, how can I write the templates from an sd card? I use an esp32 and the libraries:

#include "FS.h"
#include "SD.h"
#include "SPI.h"

I have a txt that stores only 1 template:

0x3, 0x3, 0x5B, 0x16, 0x14, 0x1, 0x3C, 0x1, 0x84, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x12, 0x0, 0x4, 0x0, 0x85, 0x0, 0x3C, 0x0, 0xC0, 0x3C, 0xCC, 0xCF, 0xFF, 0xF3, 0xFE, 0xFF, 0xFF, 0xBE, 0xEB, 0xAA, 0xBA, 0xAA, 0xAA, 0xAA, 0xBA, 0xA9, 0x95, 0x65, 0x55, 0x55, 0x55, 0x55, 0x51, 0x15, 0x54, 0x0, 0x0, 0x0, 0x44, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x57, 0x12, 0x5A, 0x5E, 0x1C, 0x99, 0x53, 0x5E, 0x56, 0x98, 0x4, 0x3E, 0x35, 0x1A, 0xD5, 0xBE, 0x67, 0x9A, 0x84, 0x3E, 0x26, 0x29, 0x53, 0xDE, 0x20, 0xAD, 0xD2, 0x1E, 0x33, 0x2E, 0x14, 0x7E, 0x76, 0xBE, 0x8, 0x1E, 0x51, 0xC0, 0xD9, 0xFE, 0x64, 0x23, 0xD9, 0xBF, 0x14, 0x26, 0xE9, 0x7F, 0x73, 0x38, 0x47, 0x7F, 0x47, 0xC0, 0x97, 0x3F, 0x27, 0xC2, 0x4E, 0x9F, 0x39, 0x3C, 0xD1, 0xFC, 0x3B, 0x3F, 0xC0, 0x98, 0x6C, 0xC4, 0xCA, 0xB8, 0x68, 0x40, 0x9C, 0xB9, 0x68, 0x43, 0x1E, 0x39, 0x68, 0xC5, 0x9E, 0x19, 0x7C, 0x21, 0x1A, 0x5E, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x3, 0x3, 0x57, 0x15, 0x0, 0x1, 0x20, 0x1, 0x85, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x5, 0x0, 0x4, 0x0, 0x7D, 0x0, 0x0, 0x0, 0x0, 0xC, 0x0, 0xCC, 0xC, 0xF3, 0xF3, 0xFF, 0xFF, 0xEF, 0xFF, 0xBF, 0xBE, 0xBA, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 0xA9, 0x95, 0x55, 0x55, 0x55, 0x54, 0x55, 0x54, 0x11, 0x54, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,

How can I write all that in the function

  `uint8_t fingerTemplate[512]={0x03,0x0E,....your template data.....};`

As I understand 
`memset(fingerTemplate, 0xff, 512); `
has something to do with it.

Thank you very much in advance :)

dody3030, 2022-09-14:
> it's better to keep your work in your repository and share the repository link in the comment section.

dody3030, 2022-09-14:
Hello everybody, I'm confused about the template size, I got R503 fingerprint when I noticed that the template size is 768 bytes not 512, is there is anything I need to change in AssifKhan991's libraries. Thanks in advance

AsifKhan991, 2022-09-14:
@dody3030 just change the array buffer size to 768. I.e : `finger.get_template_buffer(768,your_arr_buff)` to get the template data and `finger.write_template_to_sensor(768, your_arr_buff)` to write template back to sensor.

dody3030, 2022-09-14:
> write_template_to_sensor

@AsifKhan991 what about the memset(ref_buf, 0xff, 512) in Adafruit_Fingerprint.cpp

AsifKhan991, 2022-09-14:
> > write_template_to_sensor
> 
> @AsifKhan991 what about the memset(ref_buf, 0xff, 512) in Adafruit_Fingerprint.cpp

No issues if left untouched. Anyways, updated this to take value from 'bufsize' variable.

AsifKhan991, 2022-09-14:
> > write_template_to_sensor
> 
> @AsifKhan991 what about the memset(ref_buf, 0xff, 512) in Adafruit_Fingerprint.cpp

Updated this to take value from 'bufsize' variable

dody3030, 2022-09-14:
The controller get crash and reboot immediately after successfully store the model finger.storeModel, so the finger is stored successfully, then the model is loaded successfully, getModel(), the crash start after calling the function finger.get_template_buffer 

ID 2
Stored!
Attempting to get template...
Transferring Template....

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:215 loop_task

>>>stack>>>

ctx: sys
sp: 3fffeee0 end: 3fffffb0 offset: 0000
3fffeee0:  4020d968 00009ba3 3ffeefa4 4020d99b  
3fffeef0:  000000fe 00000000 00000000 00000000  
3fffef00:  00000000 00000000 00000000 3ffeea78  
3fffef10:  3ffeea98 3ffeea98 3ffee450 3ffeea34  
3fffef20:  3fffdad0 00000000 3ffe85e4 40203f02  
3fffef30:  40236b02 3fffdab0 00000000 40203f6f  
3fffef40:  00000000 3fffdad0 3ffeea34 402038fc  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 00044f11 00000000 00000005  
3fffef70:  00000000 aa55aa55 000000d4 40105951  
3fffef80:  40105957 00000000 00000005 40100f08  
3fffef90:  4010000d 30602020 00044f11 401000ac  
3fffefa0:  40226e40 3fffef3c 40226df9 3ffffd28  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feef
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:215 loop_task

>>>stack>>>

AsifKhan991, 2022-09-14:
@dody3030 need to see your code

dody3030, 2022-09-14:
@AsifKhan991  I'm using the default example, I just added the changes at the end of the function.

### > uint8_t getFingerprintEnroll() {
> uint8_t f_buf[512]; //here is where the template data is sotred
> int s = -1;
> Serial.print("Waiting for valid finger to enroll as #"); Serial.println(id);
> finger.LEDcontrol(FINGERPRINT_LED_FLASHING, 25, FINGERPRINT_LED_PURPLE, 0);
> while (s != FINGERPRINT_OK) {
> s = finger.getImage();
> switch (s) {
> case FINGERPRINT_OK:
> Serial.println("Image taken");
> finger.LEDcontrol(FINGERPRINT_LED_FLASHING, 25, FINGERPRINT_LED_PURPLE, 1);
> delay(2000);
> break;
> case FINGERPRINT_NOFINGER:
> // Serial.println(".");
> break;
> case FINGERPRINT_PACKETRECIEVEERR:
> Serial.println("Communication error");
> break;
> case FINGERPRINT_IMAGEFAIL:
> Serial.println("Imaging error");
> break;
> default:
> Serial.println("Unknown error");
> break;
> }
> }
> 
> // OK success!
> 
> s = finger.image2Tz(1);
> switch (s) {
> case FINGERPRINT_OK:
> Serial.println("Image converted");
> break;
> case FINGERPRINT_IMAGEMESS:
> Serial.println("Image too messy");
> return s;
> case FINGERPRINT_PACKETRECIEVEERR:
> Serial.println("Communication error");
> return s;
> case FINGERPRINT_FEATUREFAIL:
> Serial.println("Could not find fingerprint features");
> return s;
> case FINGERPRINT_INVALIDIMAGE:
> Serial.println("Could not find fingerprint features");
> return s;
> default:
> Serial.println("Unknown error");
> return s;
> }
> 
> Serial.println("Remove finger");
> finger.LEDcontrol(FINGERPRINT_LED_FLASHING, 25, FINGERPRINT_LED_BLUE, 0);
> //////////////// delay(2000);
> s = 0;
> while (s != FINGERPRINT_NOFINGER) {
> s = finger.getImage();
> }
> Serial.print("ID "); Serial.println(id);
> s = -1;
> Serial.println("Place same finger again");
> finger.LEDcontrol(FINGERPRINT_LED_FLASHING, 25, FINGERPRINT_LED_PURPLE, 0);
> while (s != FINGERPRINT_OK) {
> s = finger.getImage();
> switch (s) {
> case FINGERPRINT_OK:
> Serial.println("Image taken");
> finger.LEDcontrol(FINGERPRINT_LED_FLASHING, 25, FINGERPRINT_LED_PURPLE, 1);
> break;
> case FINGERPRINT_NOFINGER:
> // Serial.print(".");
> break;
> case FINGERPRINT_PACKETRECIEVEERR:
> Serial.println("Communication error");
> break;
> case FINGERPRINT_IMAGEFAIL:
> Serial.println("Imaging error");
> break;
> default:
> Serial.println("Unknown error");
> break;
> }
> }
> 
> // OK success!
> 
> s = finger.image2Tz(2);
> switch (s) {
> case FINGERPRINT_OK:
> Serial.println("Image converted");
> break;
> case FINGERPRINT_IMAGEMESS:
> Serial.println("Image too messy");
> return s;
> case FINGERPRINT_PACKETRECIEVEERR:
> Serial.println("Communication error");
> return s;
> case FINGERPRINT_FEATUREFAIL:
> Serial.println("Could not find fingerprint features");
> return s;
> case FINGERPRINT_INVALIDIMAGE:
> Serial.println("Could not find fingerprint features");
> return s;
> default:
> Serial.println("Unknown error");
> return s;
> }
> 
> // OK converted!
> Serial.print("Creating model for #");  Serial.println(id);
> 
> s = finger.createModel();
> if (s == FINGERPRINT_OK) {
> Serial.println("Prints matched!");
> } else if (s == FINGERPRINT_PACKETRECIEVEERR) {
> Serial.println("Communication error");
> return s;
> } else if (s == FINGERPRINT_ENROLLMISMATCH) {
> Serial.println("Fingerprints did not match");
> newfingerprint() ;
> // return s;
> } else {
> Serial.println("Unknown error");
> return s;
> }
> 
> Serial.print("ID "); Serial.println(id);
> s = finger.storeModel(id);
> if (s == FINGERPRINT_OK) {
> //id = (finger.templateCount + 1);
> Serial.println("Stored!");
> finger.LEDcontrol(FINGERPRINT_LED_FLASHING, 25, FINGERPRINT_LED_BLUE, 20);
> finger.LEDcontrol(FINGERPRINT_LED_GRADUAL_ON, 200, FINGERPRINT_LED_BLUE);
> delay(2000);
> finger.LEDcontrol(FINGERPRINT_LED_GRADUAL_OFF, 200, FINGERPRINT_LED_BLUE);
> delay(2000);
> finger.LEDcontrol(FINGERPRINT_LED_GRADUAL_ON, 200, FINGERPRINT_LED_BLUE);
> delay(2000);
> finger.LEDcontrol(FINGERPRINT_LED_GRADUAL_OFF, 200, FINGERPRINT_LED_BLUE);
> delay(2000);
> } else if (s == FINGERPRINT_PACKETRECIEVEERR) {
> Serial.println("Communication error");
> return s;
> } else if (s == FINGERPRINT_BADLOCATION) {
> Serial.println("Could not store in that location");
> return s;
> } else if (s == FINGERPRINT_FLASHERR) {
> Serial.println("Error writing to flash");
> return s;
> } else {
> Serial.println("Unknown error");
> return s;
> }
> 
> // Try to get the template //
> Serial.println("Attempting to get template...");
> if (finger.getModel() == FINGERPRINT_OK) {  //requesting sensor to transfer the template data to upper computer (this microcontroller)
> Serial.println("Transferring Template....");
> } else {
> Serial.println("Failed to transfer template");
> return s;
> }
> 
> 
> if (finger.get_template_buffer(768, f_buf) == FINGERPRINT_OK) { //read the template data from sensor and save it to buffer f_buf
> Serial.println("Template data (comma sperated HEX):");
> for (int k = 0; k < (512 / finger.packet_len); k++) { //printing out the template data in seperate rows, where row-length = packet_length
> for (int l = 0; l < finger.packet_len; l++) {
> Serial.print("0x");
> Serial.print(f_buf[(k * finger.packet_len) + l], HEX);
> Serial.print(",");
> }
> }
> }
> // End getting template.
> delay(2000);
> return true;
> }

AsifKhan991, 2022-09-14:
@dody3030  issues with your code:

1) You are using `f_buf[512]` when you mentioned your template size is 768 bytes. So replace all 512 to 768 in your code

2) Did not call the `finger.loadModel(id)` before calling `finger.getModel()`. Refer to "show_saved_template.ino" example.

dody3030, 2022-09-14:
> @dody3030 issues with your code:
> 
>     1. You are using `f_buf[512]` when you mentioned your template size is 768 bytes. So replace all 512 to 768 in your code
> 
>     2. Did not call the `finger.loadModel(id)` before calling `finger.getModel()`. Refer to "show_saved_template.ino" example.

Actually am trying to get the hex values during the enrollment, so do I need to loadModel() before I getModel?? I checked your previous code examples trying to understand in which stage exactly I have to get the hex, taking into account the template size is different, I will use your example get_direct_template. thank you for your replies really appreciated.

AsifKhan991, 2022-09-14:
> Actually am trying to get the hex values during the enrollment, so do I need to loadModel() before I getModel??

Usually the template remains on the char buffer if no ther operation's done after enrollment. But it might be otherwise due to model variation. So yeah, safe to do so.

dody3030, 2022-09-14:
@AsifKhan991 from the example you have shared before, store_template_to_buf ,, I added only else statement to check if the function is executed successfully or not, I changed all 512 to 768 but not working.

uint8_t f_buf[768]; //here is where the template data is sotred

### void store_template_to_buf(){

  Serial.println("Waiting for valid finger....");
  while (finger.getImage() != FINGERPRINT_OK) { // press down a finger take 1st image 
  }
  Serial.println("Image taken");

  if (finger.image2Tz(1) == FINGERPRINT_OK) { //creating the charecter file for 1st image 
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }

  Serial.println("Remove finger");
  delay(2000);
  uint8_t p = 0;
  while (p != FINGERPRINT_NOFINGER) {
    p = finger.getImage();
  }

  Serial.println("Place same finger again, waiting....");
  while (finger.getImage() != FINGERPRINT_OK) { // press the same finger again to take 2nd image
  }
  Serial.println("Image taken");

  if (finger.image2Tz(2) == FINGERPRINT_OK) { //creating the charecter file for 2nd image 
    Serial.println("Image converted");
  } else {
    Serial.println("Conversion error");
    return;
  }

  Serial.println("Creating model...");

  if (finger.createModel() == FINGERPRINT_OK) {  //creating the template from the 2 charecter files and saving it to char buffer 1
    Serial.println("Prints matched!");
    Serial.println("Template created");
  } else {
    Serial.println("Template not build");
    return;
  }

  Serial.println("Attempting to get template..."); 
  if (finger.getModel() == FINGERPRINT_OK) {  //requesting sensor to transfer the template data to upper computer (this microcontroller)
    Serial.println("Transferring Template...."); 
  } else {
    Serial.println("Failed to transfer template");
    return;
  }
  
  if (finger.get_template_buffer(768, f_buf) == FINGERPRINT_OK) { //read the template data from sensor and save it to buffer f_buf
    Serial.println("Template data (comma sperated HEX):");
    for (int k = 0; k < (768/finger.packet_len); k++) { //printing out the template data in seperate rows, where row-length = packet_length
      for (int l = 0; l < finger.packet_len; l++) {
        Serial.print("0x");
        Serial.print(f_buf[(k * finger.packet_len) + l], HEX);
        Serial.print(",");
      }
      Serial.println("");
    }
  }else{
    Serial.println("Failed to get the template into buffer..!!");
  }

}

AsifKhan991, 2022-09-14:
> @AsifKhan991 from the example you have shared before, store_template_to_buf ,, I added only else statement to check if the function is executed successfully or not, I changed all 512 to 768 but not working.

There was another issue with the function. Fixed it, can you check?

[33 more replies]