adafruit/Adafruit-Fingerprint-Sensor-Library #41: Copy whole Template Database from one sensor to another sensor...???

TehseenHasan, 2018-03-16:
please add a function to copy complete fingerprint database from one sensor to other sensor...

one-player, 2018-04-03:
This is very necessary. If two or more sensors are used, this function would be very useful

ladyada, 2018-09-04:
is this a school or work assignment? why are so many people asking for this function?

ladyada, 2018-09-05:
common requirement for *what* - a product? homework assignment?

TehseenHasan, 2018-09-07:
a project... or many projects .. like a gate through which members go through in and out, etc

ladyada, 2018-09-07:
you still need to register all their fingerprints though, so why would you have to move the template?

TehseenHasan, 2018-09-08:
Register only in one sensor than copy the template to other sensor(s) is best way instead of enrolling same person in different sensors.

ladyada, 2018-09-08:
what fingerprint sensor did you buy, the url? thx

TehseenHasan, 2018-09-08:
I use this (R305) Sensor: https://hallroad.org/product/r305-fingerprint-scanner-module-in-pakistan/

and also used this (FPM10A) with your library: https://hallroad.org/product/fingerprint-sensor-fpm10a/

both works very good with this library...

ladyada, 2018-09-08:
great! please email hallroad and ask them to add this feature support

TehseenHasan, 2018-09-08:
wow What a great suggestion by the way ...!!!  Why don't you tell me before ...  Than can you tell me what is the use of you library..?

ladyada, 2018-09-08:
ok thanks! please let us know what they say :)

dhalbert, 2018-09-08:
I think there's a potential security issue, since you could access or steal a sensor and then download the fingerprint profiles to spoof the same fingerprint elsewhere.

cyrux03, 2018-09-17:
I have 19 fingerprints sensor, but the task is copy the same database in all sensors

cyrux03, 2018-09-25:
nothing, I still waiting for your help.

cyrux03, 2018-09-26:
????, why you delete my comment, I have 19 adafruit sensors, I bought the sensors from a reseller in my country, but they are adafruit, model fpm10a

ladyada, 2018-09-26:
@cyrux03 please don't bump this thread, it wont get you an answer any sooner - we dont know of how to do this task that people keep asking for :)

milindrc, 2018-09-27:
Please add a function for storing a extracted template, or copying the database.

closed 2019-08-30 by ladyada