adafruit/Adafruit-Fingerprint-Sensor-Library #58: Which serial port does esp8266 use？

phonixivan, 2019-05-29:
I use a lot of serial ports on esp8266, but “Did not find fingerprint sensor :(”，（use usb port on computer is normal, so the sensor can work properly ), So is esp8266 supported?

hagre, 2019-06-21:
esp8266 is working. I use wemos d1 mini on Pin D2 and D3. With PlatformIO you can set D2 and D3 directly into softserial. At Arduino IDE you should set 4 and 0 to get the same effects. 
Additional hint for me was to set the hardwareserial to 115200baud in order to avoid some WDT resets.

shnkreddy98, 2019-07-17:
> I use a lot of serial ports on esp8266, but “Did not find fingerprint sensor :(”，（use usb port on computer is normal, so the sensor can work properly ), So is esp8266 supported?

Dis you find a fix? I'm getting the same error no matter what ports i use.

jimmygyl, 2019-08-10:
SoftwareSerial mySerial(D7, D8, false, 256);
Adafruit_Fingerprint finger = Adafruit_Fingerprint(&mySerial);

void setup(){
Serial.begin(115200);
  Serial.println("Fingerprint template extractor");

  // set the data rate for the sensor serial port
  finger.begin(57600);
  
  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1);
  }
}

Try this..

closed 2019-08-30 by ladyada