adafruit/Adafruit-Fingerprint-Sensor-Library #76: Minor tweak for better UI

Alexandros-Kordatzakis, 2020-05-08:
Not exactly an issue, rather than a minor tweak for better UI. When runing the example "fingerprint", when there are no fingerprints stored it still says the same message as when there are 1-XXX saved ID's. I wrote this code so that when there are no data stored (0 Finger ID's) it displays the message:
**_Sensor doesn't contain any fingerprint data. Please run the 'enroll' example._**.
Just for better UI...

Here is the code:

`finger.getTemplateCount();

    if (finger.templateCount == 0) {
    Serial.print("Sensor doesn't contain any fingerprint data. Please run the 'enroll' example.");
  } 
  else {
    Serial.println("Waiting for valid finger...");
      Serial.print("Sensor contains "); Serial.print(finger.templateCount); Serial.println(" templates");
  }`

it's from line 49, for the `finger.getTemplateCount();`

Edited **fingerprint.ino** file:  https://hastebin.com/bizotababe.cpp

Thank you.

ladyada, 2020-05-08:
hi you can submit a PR for inclusion!

Alexandros-Kordatzakis, 2020-05-08:
Copy that!

closed 2020-05-08 by Alexandros-Kordatzakis