adafruit/Adafruit-Fingerprint-Sensor-Library #82: Examples won't work with MEGA

caternuson, 2020-08-19:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=168325&p=824485#p824485

For a MEGA - current preprocessor logic in example(s) will setup a software serial instance on unsupported pins. Since conditionals are being used, can't simply "manually" set to HW serial, which is what the guide has you do:
https://learn.adafruit.com/adafruit-optical-fingerprint-sensor/wiring-for-use-with-arduino#soft-and-hard-serial-2976065-15

ladyada, 2020-08-19:
oof should be using hardware serial - mega has tons of serial ports :)

caternuson, 2020-08-20:
Got a preference between tweaking the preprocessor logic vs. no logic and just commented out code?

ladyada, 2020-08-20:
confused... why wouldnt it just work with Serial1? 
https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/master/examples/enroll/enroll.ino#L30

caternuson, 2020-08-20:
Oh, it would. But for a MEGA, the existing preprocessor logic will land you here:
https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/master/examples/enroll/enroll.ino#L25

So either need to add some more layers to the logic (use `__AVR_ATmega2560__`). Or just remove it and rely on commenting / uncommenting as needed (which is what the guide has you do).

ladyada, 2020-08-20:
oh oh duh i forgot the mega is an avr. yes can you add a check to !defined() for the mega?

caternuson, 2020-08-20:
Fixed with #83

closed 2020-08-20 by caternuson