adafruit/Adafruit-Fingerprint-Sensor-Library #88: Template Count = 0 when templates exist

jshep321, 2020-09-29:
- Arduino board:  HUZZAH32
- Arduino IDE version 1.8.12
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

Hi,
The output from reading sensor parameters works fine, but the Template Count is always 0.

11:47:51.015 -> Reading sensor parameters...
11:47:51.050 -> Status: 0x4
11:47:51.050 -> Sys ID: 0x0
11:47:51.050 -> Capacity: 200
11:47:51.050 -> Security level: 3
11:47:51.050 -> Device address: FFFFFFFF
11:47:51.050 -> Packet len: 128
11:47:51.050 -> Baud rate: 57600
11:47:51.050 -> Template count:0

ladyada, 2020-09-29:
hi what is the exact sensor you are using?

jshep321, 2020-10-02:
Ah sorry...  R503 (Product ID 4651).

ladyada, 2020-10-02:
ok it may not support that report. does it work otherwise

jshep321, 2020-10-02:
yes, works quite well thanks! 

From the hzgrow datasheet (ver 1.1) this appears to be supported... assuming I'm interpreting the "finger library size" correctly.  
[image]

caternuson, 2024-01-25:
It's not clear what example sketch was being run here. The only example that uses `getTemplateCount()` is the `fingerprint` example, but it should produce different output than what is shown above:
https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/c1274756159b94fb8042900824ab6dd3738449a7/examples/fingerprint/fingerprint.ino#L65-L74

Running a modified version of the `enroll` sketch that adds these lines to the end of `setup()`:
```cpp
  finger.getTemplateCount();
  Serial.print(F("Template count: ")); Serial.println(finger.templateCount);
```
produces output with expected results:
```
Adafruit Fingerprint sensor enrollment
Found fingerprint sensor!
Reading sensor parameters
Status: 0x4
Sys ID: 0x0
Capacity: 200
Security level: 3
Device address: FFFFFFFF
Packet len: 128
Baud rate: 57600
Template count: 2
Ready to enroll a fingerprint!
Please type in the ID # (from 1 to 127) you want to save this finger as...
```

**Please provide complete code listing for an example that demonstrates this issue.**