adafruit/Adafruit-Fingerprint-Sensor-Library #104: Library compatible with PID 4750 sensor?

caternuson, 2022-02-07:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=187741
Using this:
https://www.adafruit.com/product/4750

Intentionally placing wrong finger on second scan but not getting mismatch error.

marcelchelo, 2023-01-23:
I second this, using this fingerprint sensor:
https://www.adafruit.com/product/4651

When purposely using different fingers during enrolling, a mismatched error is expected but it does not occur.

atomdog, 2024-01-23:
> I second this, using this fingerprint sensor: https://www.adafruit.com/product/4651
> 
> When purposely using different fingers during enrolling, a mismatched error is expected but it does not occur.

I also have this problem.

caternuson, 2024-01-24:
I can reliably recreate this behavior with PID 4750 mentioned in initial post above.

But it's working OK testing with PID 4651.

Here's with the **same finger**, just to prove it does match as expected:
```
Adafruit Fingerprint sensor enrollment
Found fingerprint sensor!
Reading sensor parameters
Status: 0x4
Sys ID: 0x0
Capacity: 200
Security level: 3
Device address: FFFFFFFF
Packet len: 128
Baud rate: 57600
Ready to enroll a fingerprint!
Please type in the ID # (from 1 to 127) you want to save this finger as...
Enrolling ID #1
Waiting for valid finger to enroll as #1
........................................................Image taken
Image converted
Remove finger
ID 1
Place same finger again
.................................................Image taken
Image converted
Creating model for #1
Prints matched!
ID 1
Stored!
Ready to enroll a fingerprint!
Please type in the ID # (from 1 to 127) you want to save this finger as...
```

And here's with using **different fingers**. Tried four times but it never saw them as same.
```
Adafruit Fingerprint sensor enrollment
Found fingerprint sensor!
Reading sensor parameters
Status: 0x4
Sys ID: 0x0
Capacity: 200
Security level: 3
Device address: FFFFFFFF
Packet len: 128
Baud rate: 57600
Ready to enroll a fingerprint!
Please type in the ID # (from 1 to 127) you want to save this finger as...
Enrolling ID #1
Waiting for valid finger to enroll as #1
...................................................................................................................Image taken
Image converted
Remove finger
ID 1
Place same finger again
.........................................Image taken
Image converted
Creating model for #1
Fingerprints did not match
Ready to enroll a fingerprint!
Please type in the ID # (from 1 to 127) you want to save this finger as...
Enrolling ID #1
Waiting for valid finger to enroll as #1
..........................................................Image taken
Image converted
Remove finger
ID 1
Place same finger again
........................................Image taken
Image converted
Creating model for #1
Fingerprints did not match
Ready to enroll a fingerprint!
Please type in the ID # (from 1 to 127) you want to save this finger as...
Enrolling ID #1
Waiting for valid finger to enroll as #1
............................................................................Image taken
Image converted
Remove finger
ID 1
Place same finger again
........................................................................Image taken
Image converted
Creating model for #1
Fingerprints did not match
Ready to enroll a fingerprint!
Please type in the ID # (from 1 to 127) you want to save this finger as...
```

caternuson, 2024-01-25:
Issue #120 is a repeat for PID 4651 behavior.

krangchen, 2024-04-28:
Is there any progress on this? :)
I'm in the same boat than the initial author...

markolino85, 2024-12-28:
Someone can try this code ? There is a modification for the security level and a manual comparison for the two images.
Just to be sure that the mismatch will be detected .

`#include <Adafruit_Fingerprint.h>
#include <HardwareSerial.h>

// Configure UART
HardwareSerial mySerial(2); // Use UART2 (GPIO17 RX, GPIO16 TX)
Adafruit_Fingerprint finger(&mySerial);

void setup() {
  Serial.begin(115200); // Serial monitor
  mySerial.begin(57600); // Sensor baud rate

  // Initialize the sensor
  if (finger.begin()) {
    Serial.println("Sensor detected!");
    finger.setSecurityLevel(5); // Set the highest security level
    Serial.println("Security level set to 5 (maximum).");
  } else {
    Serial.println("Error: Unable to communicate with the sensor.");
    while (1);
  }
}

void loop() {
  Serial.println("Place your finger on the sensor...");
  if (getFingerprintEnroll() == FINGERPRINT_OK) {
    Serial.println("Fingerprint successfully saved!");
  } else {
    Serial.println("Error during enrollment. Please try again.");
  }
  delay(5000); // Wait a bit before retrying
}

uint8_t getFingerprintEnroll() {
  int id = 1; // Unique ID for each fingerprint
  Serial.print("Enrolling fingerprint with ID: ");
  Serial.println(id);

  // Step 1: Capture the first image
  while (finger.getImage() != FINGERPRINT_OK) {
    Serial.println("Place your finger correctly on the sensor.");
    delay(1000);
  }
  if (finger.image2Tz(1) != FINGERPRINT_OK) {
    Serial.println("Error converting the first image.");
    return FINGERPRINT_ERR;
  }
  Serial.println("First image captured. Remove your finger.");
  delay(2000);

  // Step 2: Capture the second image
  while (finger.getImage() != FINGERPRINT_NOFINGER);
  Serial.println("Place the same finger again.");
  delay(2000);

  while (finger.getImage() != FINGERPRINT_OK) {
    Serial.println("Place your finger correctly on the sensor.");
    delay(1000);
  }
  if (finger.image2Tz(2) != FINGERPRINT_OK) {
    Serial.println("Error converting the second image.");
    return FINGERPRINT_ERR;
  }
  Serial.println("Second image captured.");

  // Step 3: Manual comparison of the images
  if (finger.compare() != FINGERPRINT_OK) {
    Serial.println("The two images do not match. Please use the same finger.");
    return FINGERPRINT_ERR;
  }
  Serial.println("The two images match.");

  // Step 4: Create the fingerprint model
  if (finger.createModel() != FINGERPRINT_OK) {
    Serial.println("Error creating the fingerprint model.");
    return FINGERPRINT_ERR;
  }
  Serial.println("Fingerprint model created successfully.");

  // Step 5: Save the model in the sensor's memory
  if (finger.storeModel(id) != FINGERPRINT_OK) {
    Serial.println("Error saving the fingerprint model.");
    return FINGERPRINT_ERR;
  }

  Serial.println("Fingerprint model saved in the sensor's memory.");
  return FINGERPRINT_OK;
}
`