adafruit/Adafruit-Fingerprint-Sensor-Library #114: Cannot use setBaudRate() to change from default 57600bps

alastaira, 2022-08-02:
Using v2.1.0 of the fingerprint library. I'm using an UNO with only one dedicated hardware UART, so I'm using SoftSerial to emulate a software serial connection to the sensor (as in the example code), and this generally works fine - I can run all the example code successfully.
However, since SoftSerial is known to be unreliable at high baud rates, I'm attempting to set the sensor from its default 57600 baud rate down to 9600 instead. So, I first connect at 57600 baud, vertify the password etc. (all of which works fine), and then issue the following:

`finger.setBaudRate(FINGERPRINT_BAUDRATE_9600);`

The command is acknowledged and return value of 1 is received from setting the register, so it appears to have been successful written. 

However, the existing serial connection seems to persist at 57600bps. And, if I power-cycle the unit and attempt to connect at 9600, the serial connection fails. I can only connect at 57600 still, and calling finger.getParameters() confirms that the unit is still running at 57600, so it appears that the command had no effect.

Is there something else I need to do to commit the register change? For comparison, I tried using finger.setSecurityLevel() to change the security level, and this works correctly - I can write any value from 1 to 5 and that persists between power cycles, and the new value is reported correctly by getParameters().

Thanks.

ladyada, 2022-08-02:
could be a regression failure. that said, softwareserial is pretty good at 57600 and we test at that speed no problem!

alastaira, 2022-08-02:
Thanks for the quick response!
I'll have a check back through and see if there have been any commits that could have caused that function to regress - from what I can see in the (minimal!) datasheet, it certainly seems like it _should_ work. I guess because there were no examples demonstrating its usage I just wondered whether there were any known issues, but I'll investigate and comment back if I find anything.

I find that 57600 baud for SoftSerial only works if you have a very simple sketch (and not much data being transmitted) - but as soon as you have any non-trivial logic in loop() you risk corrupted data. Part of this sketch involves transferring the library of fingerprint templates, so I'd prefer it to be robust. I could switch to AltSoftSerial, which is more reliable at higher speeds, but thought I'd try to drop to 9600 as it seemed like it should be easy to do! ;)

frami, 2022-09-03:
The following PR #115 solve the problem.

alastaira, 2022-09-09:
> The following PR #115 solve the problem.

Oh, good spot - can't believe I missed that! Thankyou :)

henfri, 2022-11-03:
Thanks @frami !
I tried it with this code:
```
    finger.begin(57600);
    Serial.println("Baudrate");
    Serial.println(finger.baud_rate);
    Serial.println("Set Baudrate. return value: (0 is ok, 1 is not ok)");
    String ret = "";
    finger.begin(9600);
    ret=finger.setBaudRate(FINGERPRINT_BAUDRATE_9600);
    Serial.println(ret);
    Serial.println("Baudrate");
    Serial.println(finger.baud_rate);
```
it returns:
```
Baudrate
57600
Set Baudrate. return value:
0
Baudrate
57600
Found fingerprint sensor!
```
So, it seems that the Baudrate remains at 57600 baud. 
What am I doing wrong?

Apart from this: what is the idear behind set_baudrate? Will it be stored in the Sensor?

The reason I ask: My cable is a bit longer.  Thus, the connection at the high baud rate may fail. I rely on the fact that the connection would always be done at 9600 baud.

Best regards,
Hendrik

alastaira, 2022-11-03:
> Thanks @frami ! I tried it with this code:
> 
> ```
>     finger.begin(57600);
>     Serial.println("Baudrate");
>     Serial.println(finger.baud_rate);
>     Serial.println("Set Baudrate. return value: (0 is ok, 1 is not ok)");
>     String ret = "";
>     ret=finger.setBaudRate(FINGERPRINT_BAUDRATE_9600);
>     Serial.println(ret);
>     Serial.println("Baudrate");
>     finger.begin(9600);
>     Serial.println(finger.baud_rate);
> ```
> 
> it returns:
> 
> ```
> Baudrate
> 57600
> Set Baudrate. return value:
> 0
> Baudrate
> 57600
> Found fingerprint sensor!
> ```
> 
> So, it seems that the Baudrate remains at 57600 baud. What am I doing wrong?
> 
> I also tried:
> 
> ```
>     finger.begin(9600);
>     Serial.println("Baudrate");
>     Serial.println(finger.baud_rate);
>     Serial.println("Set Baudrate. return value: (0 is ok, 1 is not ok)");
>     String ret = "";
>     ret=finger.setBaudRate(FINGERPRINT_BAUDRATE_9600);
>     finger.begin(FINGERPRINT_BAUDRATE_9600);
>     Serial.println(ret);
>     Serial.println("Baudrate");
>     Serial.println(finger.baud_rate);
> ```
> 
> and get
> 
> ```
> Baudrate
> 57600
> Set Baudrate. return value: (0 is ok, 1 is not ok)
> 0
> Baudrate
> 57600
> ```
> 
> Apart from this: what is the idear behind set_baudrate? Will it be stored in the Sensor?
> 
> The reason I ask: My cable is a bit longer. Thus, the connection at the high baud rate may fail. I rely on the fact that the connection would always be done at 9600 baud.
> 
> Best regards, Hendrik

The PR has not been implemented into the master branch yet, so did you apply the changes yourself?

henfri, 2022-11-04:
Hello,

this was the problem:
```
    ret=finger.setBaudRate(FINGERPRINT_BAUDRATE_9600);
    finger.begin(FINGERPRINT_BAUDRATE_9600);
```
`FINGERPRINT_BAUDRATE_9600 `is `0x1` and used to SET the baudrate.
To connect, you need `finger.begin(9600)`.

Nevertheless, I am a bit confused now.... 

```
    finger.begin(9600);
    Serial.println("Baudrate");
    Serial.println(finger.baud_rate);
```
returns 

```
Baudrate
57600
```

I finally found out, that 
`finger.getParameters();`
needs to be called, before the baudrate is correctly reported.

Is that the intended behaviour? I would not expect that a wrong value is returned, if I do not call getParameters first.
I would expect "None" or so.

Anyway, for me it works now :-)

Please include the PR in the next release!

caternuson, 2024-01-25:
Closing. Should be fixed by #115.

closed 2024-01-25 by caternuson