adafruit/Adafruit-Fingerprint-Sensor-Library #118: Fingerprint sensor constantly flashing.

nsafai, 2022-10-06:
Is it possible to get the fingerprint sensor to stop flashing incessantly? e.g. it only flashes once it detects the presence of a fingerprint on the sensor? The datasheet also mentions "finger detection" in regards to the 2uA standby current. 

All of the sample code works _except_ the LED control sample code: https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/master/examples/ledcontrol/ledcontrol.ino which makes me wonder if that portion of the library is somehow broken?

- Arduino board:  Arduino Uno

- Arduino IDE version: Arduino 1.8.14

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
The following sketch is broken and does not control the fingerprint sensor. Again, every other sample sketch works perfectly except this one.
    - Sketch: https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/master/examples/ledcontrol/ledcontrol.ino
    - Sensor: https://smile.amazon.com/gp/product/B07BQ9VNWR/ref=ask_ql_qh_dp_hza

RobMel98, 2022-10-11:
I've got the same problem, just with a different sensor. Mine is this one: https://www.adafruit.com/product/4750.

If I use for example the _enroll_ sketch, the sensor is fadding in and out with a blue color and if a stored fingerprint is recognized it lights up in green and after a few seconds jumps back to the blue fading.

But if I upload the _ledcontrol_ sketch the sensor is just doing it's blue fading but it does not react to the color commands of that sketch.

My Arduino board is also an Arduino Uno and I'm using the Arduino IDE version 1.8.19

bdh8, 2022-11-07:
RobMel98: take a look at my response to https://github.com/adafruit/Adafruit_CircuitPython_Fingerprint/issues/48  This module (adafruit 4750) does not respond to LED commands, but can be put in "stand-by" mode with a couple extra components which turns off the breathing blue LED and has a touch sensor (White and blue leads) that can be used to tell you when to take it out of stand-by mode.

nsafai: from what I can see on Amazon at the link you gave, your sensor may work the same with Vtouch and Sout being the leads for the touch sensor (blue and yellow that the images never connect to anything).

nsafai, 2022-11-07:
@bdh8 thanks! I appreciate the followup on this. I ended up getting the round fingerprint sensor adafruit recommends on their site and the LED code works fine now. With the new sensor I was able to get LED to flash only when it's scanning a fingerprint, it's a bit quality of life upgrade. Highly recommend it!

caternuson, 2024-01-25:
Closing. Seems resolved. The [PID 4651](https://www.adafruit.com/product/4651) sensor is the only one that currently has actual LED control.

closed 2024-01-25 by caternuson