adafruit/Adafruit-Fingerprint-Sensor-Library #120: Incorrectly matched fingerprints for: Rugged Panel Mount Fingerprint Sensor with Bi-Color LED Ring - R503

marcelchelo, 2023-01-23:
If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **mkr 1500, mkr1400 and UNO**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.19**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

Running the Enroll Sketch from the Adafruit Fingerprint Sensor library.

Using the https://www.adafruit.com/product/4651  fingerprint sensor. 

When prompted to place finger on sensor, the index finger is chosen. Then you are prompted to place the same finger again to confirm.   This time however I placed a different finger (thumb) 
.
The console returns a Match,   when a  no-match is expected.      (Same with every DISTINCT finger pair)

I'd like to point out that I have https://www.adafruit.com/product/751 fingerprint sensor as well. 
Using this fingerprint model a mismatched fingerprint is accurately detected.

---------------- Enroll - Library Code Below------------------
/***************************************************
  This is an example sketch for our optical Fingerprint sensor

  Designed specifically to work with the Adafruit BMP085 Breakout
  ----> http://www.adafruit.com/products/751

  These displays use TTL Serial to communicate, 2 pins are required to
  interface
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  BSD license, all text above must be included in any redistribution
 ****************************************************/

#include <Adafruit_Fingerprint.h>

#if (defined(__AVR__) || defined(ESP8266)) && !defined(__AVR_ATmega2560__)
// For UNO and others without hardware serial, we must use software serial...
// pin #2 is IN from sensor (GREEN wire)
// pin #3 is OUT from arduino  (WHITE wire)
// Set up the serial port to use softwareserial..
SoftwareSerial mySerial(2, 3);

#else
// On Leonardo/M0/etc, others with hardware serial, use hardware serial!
// #0 is green wire, #1 is white
#define mySerial Serial1

#endif

Adafruit_Fingerprint finger = Adafruit_Fingerprint(&mySerial);

uint8_t id;

void setup()
{
  Serial.begin(9600);
  while (!Serial);  // For Yun/Leo/Micro/Zero/...
  delay(100);
  Serial.println("\n\nAdafruit Fingerprint sensor enrollment");

  // set the data rate for the sensor serial port
  finger.begin(57600);

  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1) { delay(1); }
  }

  Serial.println(F("Reading sensor parameters"));
  finger.getParameters();
  Serial.print(F("Status: 0x")); Serial.println(finger.status_reg, HEX);
  Serial.print(F("Sys ID: 0x")); Serial.println(finger.system_id, HEX);
  Serial.print(F("Capacity: ")); Serial.println(finger.capacity);
  Serial.print(F("Security level: ")); Serial.println(finger.security_level);
  Serial.print(F("Device address: ")); Serial.println(finger.device_addr, HEX);
  Serial.print(F("Packet len: ")); Serial.println(finger.packet_len);
  Serial.print(F("Baud rate: ")); Serial.println(finger.baud_rate);
}

uint8_t readnumber(void) {
  uint8_t num = 0;

  while (num == 0) {
    while (! Serial.available());
    num = Serial.parseInt();
  }
  return num;
}

void loop()                     // run over and over again
{
  Serial.println("Ready to enroll a fingerprint!");
  Serial.println("Please type in the ID # (from 1 to 127) you want to save this finger as...");
  id = readnumber();
  if (id == 0) {// ID #0 not allowed, try again!
     return;
  }
  Serial.print("Enrolling ID #");
  Serial.println(id);

  while (!  getFingerprintEnroll() );
}

uint8_t getFingerprintEnroll() {

  int p = -1;
  Serial.print("Waiting for valid finger to enroll as #"); Serial.println(id);
  while (p != FINGERPRINT_OK) {
    p = finger.getImage();
    switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image taken");
      break;
    case FINGERPRINT_NOFINGER:
      Serial.println(".");
      break;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Communication error");
      break;
    case FINGERPRINT_IMAGEFAIL:
      Serial.println("Imaging error");
      break;
    default:
      Serial.println("Unknown error");
      break;
    }
  }

  // OK success!

  p = finger.image2Tz(1);
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image converted");
      break;
    case FINGERPRINT_IMAGEMESS:
      Serial.println("Image too messy");
      return p;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Communication error");
      return p;
    case FINGERPRINT_FEATUREFAIL:
      Serial.println("Could not find fingerprint features");
      return p;
    case FINGERPRINT_INVALIDIMAGE:
      Serial.println("Could not find fingerprint features");
      return p;
    default:
      Serial.println("Unknown error");
      return p;
  }

  Serial.println("Remove finger");
  delay(2000);
  p = 0;
  while (p != FINGERPRINT_NOFINGER) {
    p = finger.getImage();
  }
  Serial.print("ID "); Serial.println(id);
  p = -1;
  Serial.println("Place same finger again");
  while (p != FINGERPRINT_OK) {
    p = finger.getImage();
    switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image taken");
      break;
    case FINGERPRINT_NOFINGER:
      Serial.print(".");
      break;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Communication error");
      break;
    case FINGERPRINT_IMAGEFAIL:
      Serial.println("Imaging error");
      break;
    default:
      Serial.println("Unknown error");
      break;
    }
  }

  // OK success!

  p = finger.image2Tz(2);
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image converted");
      break;
    case FINGERPRINT_IMAGEMESS:
      Serial.println("Image too messy");
      return p;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Communication error");
      return p;
    case FINGERPRINT_FEATUREFAIL:
      Serial.println("Could not find fingerprint features");
      return p;
    case FINGERPRINT_INVALIDIMAGE:
      Serial.println("Could not find fingerprint features");
      return p;
    default:
      Serial.println("Unknown error");
      return p;
  }

  // OK converted!
  Serial.print("Creating model for #");  Serial.println(id);

  p = finger.createModel();
  if (p == FINGERPRINT_OK) {
    Serial.println("Prints matched!");
  } else if (p == FINGERPRINT_PACKETRECIEVEERR) {
    Serial.println("Communication error");
    return p;
  } else if (p == FINGERPRINT_ENROLLMISMATCH) {
    Serial.println("Fingerprints did not match");
    return p;
  } else {
    Serial.println("Unknown error");
    return p;
  }

  Serial.print("ID "); Serial.println(id);
  p = finger.storeModel(id);
  if (p == FINGERPRINT_OK) {
    Serial.println("Stored!");
  } else if (p == FINGERPRINT_PACKETRECIEVEERR) {
    Serial.println("Communication error");
    return p;
  } else if (p == FINGERPRINT_BADLOCATION) {
    Serial.println("Could not store in that location");
    return p;
  } else if (p == FINGERPRINT_FLASHERR) {
    Serial.println("Error writing to flash");
    return p;
  } else {
    Serial.println("Unknown error");
    return p;
  }

  return true;
}

caternuson, 2024-01-25:
This behavior was also mentioned in this other issue thread:
https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/issues/104#issuecomment-1401008427

However, it can not be repeated locally using the PID 4651 sensor:
https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/issues/104#issuecomment-1908886511

markolino85, 2024-12-28:
Someone can try this code ? There is a modification for the security level and a manual comparison for the two images.
Just to be sure that the mismatch will be detected .
Hope this helps.

```
#include <Adafruit_Fingerprint.h>
#include <HardwareSerial.h>

// Configure UART
HardwareSerial mySerial(2); // Use UART2 (GPIO17 RX, GPIO16 TX)
Adafruit_Fingerprint finger(&mySerial);

void setup() {
  Serial.begin(115200); // Serial monitor
  mySerial.begin(57600); // Sensor baud rate

  // Initialize the sensor
  if (finger.begin()) {
    Serial.println("Sensor detected!");
    finger.setSecurityLevel(5); // Set the highest security level
    Serial.println("Security level set to 5 (maximum).");
  } else {
    Serial.println("Error: Unable to communicate with the sensor.");
    while (1);
  }
}

void loop() {
  Serial.println("Place your finger on the sensor...");
  if (getFingerprintEnroll() == FINGERPRINT_OK) {
    Serial.println("Fingerprint successfully saved!");
  } else {
    Serial.println("Error during enrollment. Please try again.");
  }
  delay(5000); // Wait a bit before retrying
}

uint8_t getFingerprintEnroll() {
  int id = 1; // Unique ID for each fingerprint
  Serial.print("Enrolling fingerprint with ID: ");
  Serial.println(id);

  // Step 1: Capture the first image
  while (finger.getImage() != FINGERPRINT_OK) {
    Serial.println("Place your finger correctly on the sensor.");
    delay(1000);
  }
  if (finger.image2Tz(1) != FINGERPRINT_OK) {
    Serial.println("Error converting the first image.");
    return FINGERPRINT_ERR;
  }
  Serial.println("First image captured. Remove your finger.");
  delay(2000);

  // Step 2: Capture the second image
  while (finger.getImage() != FINGERPRINT_NOFINGER);
  Serial.println("Place the same finger again.");
  delay(2000);

  while (finger.getImage() != FINGERPRINT_OK) {
    Serial.println("Place your finger correctly on the sensor.");
    delay(1000);
  }
  if (finger.image2Tz(2) != FINGERPRINT_OK) {
    Serial.println("Error converting the second image.");
    return FINGERPRINT_ERR;
  }
  Serial.println("Second image captured.");

  // Step 3: Manual comparison of the images
  if (finger.compare() != FINGERPRINT_OK) {
    Serial.println("The two images do not match. Please use the same finger.");
    return FINGERPRINT_ERR;
  }
  Serial.println("The two images match.");

  // Step 4: Create the fingerprint model
  if (finger.createModel() != FINGERPRINT_OK) {
    Serial.println("Error creating the fingerprint model.");
    return FINGERPRINT_ERR;
  }
  Serial.println("Fingerprint model created successfully.");

  // Step 5: Save the model in the sensor's memory
  if (finger.storeModel(id) != FINGERPRINT_OK) {
    Serial.println("Error saving the fingerprint model.");
    return FINGERPRINT_ERR;
  }

  Serial.println("Fingerprint model saved in the sensor's memory.");
  return FINGERPRINT_OK;
}
```