adafruit/Adafruit-Fingerprint-Sensor-Library #122: More colors for LED color control

alve89, 2023-02-09:
In addition to the existing LED color control I'd like to share more available colors on newer models (I'm using a R503 module). Maybe this is helpful for others, so just add the following lines after `#define FINGERPRINT_LED_GRADUAL_OFF 0x06 //!< Gradually off` in `Adafruit_Fingerprint.h`:

```
#define FINGERPRINT_LED_WARMWHITE 0x00  //!< warm white
#define FINGERPRINT_LED_GREEN 0x04  //!< green
#define FINGERPRINT_LED_YELLOW 0x05  //!< yellow
#define FINGERPRINT_LED_CYAN 0x06  //< cyan
#define FINGERPRINT_LED_ COLDWHITE 0x07  //< cold white
```

And as "issue": In [line 89 of Adafruit_Fingerprint.h](https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/358c6c9e439292923cace27128cd4d4191ed1ef4/Adafruit_Fingerprint.h#L89) stands `//!< Purple LEDpassword` which is obviously wrong, so please just remove the `password`.

Have a nice day all together!

caternuson, 2024-01-25:
How was `0x00` for warm white determined?

[image]

caternuson, 2024-01-25:
This also may not be available on all models depending on what firmware they are running:
[image]

h0t5tuff, 2024-02-28:
fixing last line in:
#define FINGERPRINT_LED_WARMWHITE 0x00  //!< warm white
#define FINGERPRINT_LED_GREEN 0x04  //!< green
#define FINGERPRINT_LED_YELLOW 0x05  //!< yellow
#define FINGERPRINT_LED_CYAN 0x06  //< cyan
#define FINGERPRINT_LED_ COLDWHITE 0x07  //< cold white
by removing an extra space, so it becomes:
#define FINGERPRINT_LED_COLDWHITE 0x07  //< cold white