adafruit/Adafruit-Fingerprint-Sensor-Library #127: Adding fingerprint template transfer functionality to the Adafruit-Fingerprint library

mahdi-yahyaie, 2023-12-10:
I need to be able to get a fingerprint template from an r502a module, save it, and then transfer it to another module with the same ID. In other words, I need to be able to transfer fingerprints from one module to another. Is this possible with this library? If not, can you help me add this functionality to the library so that I and others can use it? I would be grateful for any help you can provide.

caternuson, 2024-01-25:
Not currently supported. The UpChar (0x08) and DownChar (0x09) commands would need to be implemented.

There is the `getModel()` function that has a simple implementation of UpChar (transfer data from fingerprint sensor to host PC).

Umarul-Faarugh-Diallo, 2025-09-10:
S'il vous plaît je veux utiliser la fonction getModelTemplate() dans mon code mais la librairie Adafruit officielle ne l'a pas encore ajouté
je veux obtenir le Template binaire des empreintes à fin de les stocker sur une base de donnée externe.