adafruit/Adafruit-Fingerprint-Sensor-Library #130: R558 Fingerprint not picked

faizannazir, 2024-06-08:
- Arduino board:  **Nano**
- Arduino IDE version (found in Arduino -> About Arduino menu):  **2.3.2**

Fingerprint sensor is not picked

Irfanislam, 2024-10-09:
AoA, it will not work look into data sheet of R558 quite a range of instruction codes are different you can make a new copy of adafruit liberary file and make changes there in both CPP and H, because no specific R558 liberay is available yet.
Also wiring of R558 is very ugly and not sequenced well. I found that Black is TX, Yellow is RX, Red is GND, White is VCC_3V3, Green is VCC_WAKEUP and Blue is TOUCH_OUT.

faizannazir, 2024-10-10:
It works with adfruit library without modification

white, pin1-VCC
blue, pin2- TOUCH, we didn't connect it
green,pin3-VCC, we connect it with white cable.
yellow,black, pin 4,5, tx rx.
red, pin6 GND.

Irfanislam, 2024-10-10:
what about its aura led did worked as well?

faizannazir, 2024-10-10:
> what about its aura led did worked as well?

No

Gior80, 2025-07-15:
Working on it I found error, standing at r558s manual, is 0x35 = illegal data. 
command are same between 503 and 558s. has anybody figured out ?

carloswbarros, 2025-08-01:
The only thing that it doesn't seem to work is the storeModel, it's returning 53 (0x35, Illegal Data)
Did someone found a solution?

Irfanislam, 2025-08-01:
Using for loop will make your code robust, shoter and decrease size as well because repeating "p = finger.image2Tz(uint16_t location);" 4 time is not professional.

zshlomi, 2025-08-03:
thanks for the info,  looped 4 times and it works 
success !!!!

Irfanislam, 2025-08-04:
> Working on it I found error, standing at r558s manual, is 0x35 = illegal data. command are same between 503 and 558s. has anybody figured out ?

Hope you got the solution as mentioned here underneath.

Farziscu, 2025-10-05:
Hey Irfanislam, thank you very much for your kind support.
Yes, I connected wires as you described but it still doesn't work.

Current shouldn't be a problem cause I am using a FT232RL converter USB to SERIAL.
Maybe sensor is already damaged now :(

Since no led turn on when I supply it, I'm trying to send only a packet to verify password.
The packet is: 0xEF;  0x1;  0xFF;  0xFF;  0xFF;  0xFF;  0x1;  0x0;  0x7;  0x13;  0x0;  0x0;  0x0;  0x0;  0x0;  0x1B;

No data are sending from black wire.